CameraKit/camerakit-android #297: Is there a workaround for disabling microphone/audio permission check?

MoreKoo, 2017-12-31:
CameraKit is wonderful! It has saved a lot of time and frustration in my current project. I'll try to find time in the future to contribute.

I have an app that does not need any video capabilities. However, CameraKit seem to check audio/microphone permissions among others. For my users this is a bit confusing to get a popup window, asking them to allow audio/microphone access. Is there any workaround to prevent this happening, since all I need is an image capture?

xp-vit, 2018-01-07:
You can use:
`app:ckPermissions="picture"`

More documentation can be found at: http://docs.camerakit.website/#/?id=ckpermissions

waliahimanshu, 2019-10-18:
Hi I am using `v1.0.0-beta3.11` to only capture images and I am getting `microphone/audio permission ` by default (turned off) in the app 
I can not use `app:ckPermissions="picture"` as this is not available anymore ?