CameraKit/camerakit-android #302: Startpreview failed for API below 21

LordFluffy, 2018-01-15:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

### Steps to Reproduce

App crashes when opening camera on phones below 21 .

### Expected Behavior

That it works

### Actual Behavior

Crashes with startpreview has failed.

Newwaygo, 2018-02-09:
I met the same bug,
How to solve?

austinkettner, 2018-03-23:
Hey @LordFluffy and @Newwaygo Can you give us specific examples of the function being called (code block), the device your using (that is running Android sub-21), and the CameraKit version (0.12.x, 0..13.x) your using?

If occurring within the simulator, that is fine too, just want some additional clarity here before we tag this and attempted to resolve. Additionally, we recommend using #318 going forward which documentation wise will improve but is our upcoming release and Long-Term Support candidate, which has address many of our previous 0.13.x and 0.12.x issues. We are no longer patching 0.12.x and 0.13.x as we focus on 1.0.0.

jproffitt, 2018-04-15:
This happened a few times to some of our users. it was API 24 on a Motorola DROID Turbo 2. Camerakit 0.13.1. This stack overflow question might be the same thing: https://stackoverflow.com/questions/7942378/android-camera-will-not-work-startpreview-fails

It appears to be happening right here: https://github.com/CameraKit/camerakit-android/blob/1c2770c98ecd72543943cf60d27d64a9529bdae7/camerakit-core/src/main/api16/com/wonderkiln/camerakit/Camera1.java#L128

Which now that I think about it, that is only supposed to run on api levels below 21 right?

emersoncloud, 2018-07-31:
This issue has been resolved in the latest release of CameraKit v1.0.0 beta1. See the PR notes here #318 and give it a try. Report back if you are having any more issues!

closed 2018-07-31 by emersoncloud