CameraKit/camerakit-android #304: when use this methord " void captureVideo(File videoFile, VideoCapturedCallback callback)"

wangchengyan, 2018-01-16:
when use this methord " void captureVideo(File videoFile, VideoCapturedCallback callback)"

   @Override
    void stopVideo() {
        synchronized (mCameraLock) {
            if (mRecording) {
                File videoFile = getVideoFile();

stopVideo   File videoFile = getVideoFile();      option the wrong videoFile 

@Override
    void captureVideo(File videoFile, VideoCapturedCallback callback) {
        synchronized (mCameraLock) {
            try {
                if (prepareMediaRecorder(videoFile)) {
                    mMediaRecorder.start();
                    mRecording = true;
                    this.mVideoCallback = callback;
                } else {
                    releaseMediaRecorder();
                }
            } catch (IOException e) {
                releaseMediaRecorder();
            } catch (RuntimeException e) {
                releaseMediaRecorder();
            }
        }
    }

    @Override
    void stopVideo() {
        synchronized (mCameraLock) {
            if (mRecording) {
                File videoFile = getVideoFile();

                try {
                    mMediaRecorder.stop();
                    if (this.mVideoCallback != null) {
                        mVideoCallback.videoCaptured(videoFile);
                        mVideoCallback = null;
                    }
                } catch (RuntimeException e) {
                    videoFile.delete();
                }

                releaseMediaRecorder();
                mRecording = false;
            }

            stop();
            start();
        }
    }

hailovigor, 2018-01-25:
I have the same problem

emersoncloud, 2018-03-23:
Hi @wangchengyan. Could you post another comment following our issue template? I'll reOpen this issue once you have. Thanks!

closed 2018-03-23 by emersoncloud