CameraKit/camerakit-android #305: Flipped video when using front camera

hydarm, 2018-01-18:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment
Android Studio

### Steps to Reproduce

(Write your steps here:)

1.Select front camera mode
2.Capture video
3.In preview - Video is flipped

### Expected Behavior

I was expecting the video to look exactly like it did in the viewfinder

### Actual Behavior

No crashes or errors, just a flipped end result.

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

boranyldrm, 2018-01-18:
The preview problem can be solved by using TextureView rather than VideoView. However, it will be better to record the video by setting the mirrored orientation. 
```
textureView.setScaleX(-1);    // solve the mirror problem

textureView.setSurfaceTextureListener(new TextureView.SurfaceTextureListener() {
    @Override
    public void onSurfaceTextureAvailable(SurfaceTexture surfaceTexture, int i, int i1) {
        mp.setDataSource(videoFile.getAbsolutePath());
        mp.setSurface(new Surface(surfaceTexture));
        mp.prepare();
        mp.setOnPreparedListener(new MediaPlayer.OnPreparedListener() {
            @Override
            public void onPrepared(MediaPlayer mediaPlayer) {
                // infinitely show video
                mp.setLooping(true);
                mp.seekTo(0);
                mp.start();
            }
        });
        mp.start();
    }

    @Override
    public void onSurfaceTextureSizeChanged(SurfaceTexture surfaceTexture, int i, int i1) {

    }

    @Override
    public boolean onSurfaceTextureDestroyed(SurfaceTexture surfaceTexture) {
        mp.stop();
        mp.release();
        return false;
    }

    @Override
    public void onSurfaceTextureUpdated(SurfaceTexture surfaceTexture) {

    }
});
```

hydarm, 2018-01-19:
@boranyldrm Will this record the video flipped (matching the viewfinder) or just preview it flipped ?

boranyldrm, 2018-01-20:
Just preview.

emersoncloud, 2018-07-30:
Thanks for the issue. We will be re-adding video functionality in an upcoming release of 1.0.0. If you still experience the flipped video issue let us know and we will take a look. Thanks!

closed 2018-07-30 by emersoncloud

santoshthorani, 2018-09-04:
@emersoncloud  I tried on Samsung J5 prime, video is flipped recorded while  preview is ok using front camera.