CameraKit/camerakit-android #309: Failed to resolve: com.wonderkiln:camerakit:0.13.1

fcafra, 2018-01-23:
I have added     
compile 'com.wonderkiln:camerakit:0.13.1' into build.gradle (App) but i have this error 
Failed to resolve: com.wonderkiln:camerakit:0.13.1

What am i doing wrong ?

austinkettner, 2018-01-23:
@fcafra You aren't doing anything wrong, we are clearing up final issues with Bintray and the above will then work. Will update this ticket soon for you.

fcafra, 2018-01-23:
good news, thanks ...  I wait   :-)

EduardoDornel, 2018-01-23:
I have same problem

shripal17, 2018-01-24:
When will it be live back again?

dwillmc, 2018-01-24:
Working out an issue with JCenter Bintray.
For the time being can all please use the repository directly if it's not working.

In your project level build.gradle you'll see a block that looks something like:

```
allprojects {
    repositories {
        jcenter()
        maven { url 'https://camerakit.bintray.com/camerakit-android-prereleases' }
    }
}
```

Just add the 2nd repository, `maven { url 'https://camerakit.bintray.com/camerakit-android-prereleases' }`

urschrei, 2018-01-24:
@dwillmc That doesn't resolve the sync error, unfortunately.

dwillmc, 2018-01-24:
What version are you trying to resolve?

urschrei, 2018-01-24:
@dwillmc Just tried again, and it's synced. Sorry about that!

mjmotani, 2018-01-25:
Try to sync with above way getting this issue
`Error:Could not create service of type TaskArtifactStateRepository using TaskExecutionServices.createTaskArtifactStateRepository().
> Unexpected lock protocol found in lock file. Expected 3, found 0.`

austinkettner, 2018-02-07:
This should all be cleared up now. Let me know if you run into any other issues resolving the desired package version.

closed 2018-02-07 by austinkettner

fcafra, 2018-02-08:
@dwillmc  @austinkettner  Thanks

caleb-allen, 2018-02-08:
Could we get some clarification on http://docs.camerakit.website for which version we should be using?

Docs site shows `0.12.0`, master README shows `0.13.1`

austinkettner, 2018-02-08:
Latest is always shown in the README, so currently `0.13.1` we have some big documentation updates coming soon.

caleb-allen, 2018-02-08:
Awesome! Looking forward to it

NareshG-Lepton, 2023-08-10:
Could not find com.wonderkiln:camerakit:0.13.1. Required by Pix

Can anyone help on it.

jzhw0130, 2023-08-15:
Could not find com.wonderkiln:camerakit:0.13.5. 
Worked well before, stopped working now.
Can anyone help on it?

deebashd, 2023-08-28:
Any update on this?

zubaer190, 2023-09-05:
Could not find com.wonderkiln:camerakit:0.13.5.

stefanbildl, 2023-09-06:
Can you please publish to mavenCentral?

NareshG-Lepton, 2023-11-08:
waiting for long time