CameraKit/camerakit-android #322: Create new istance without preview

daniele93dc, 2018-02-11:
### Is this a bug report?

Yes 

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

### Steps to Reproduce

(Write your steps here:)

1. Create a new CameraView instance with cameraview = new CameraView(this)
2. start camera
3. take a photo

### Expected Behavior

get a photo

### Actual Behavior

Crash error: FATAL EXCEPTION: CameraViewWorker
after onImage event the app crash

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

valterh4ck3r, 2018-02-24:
What your device ?  manufacturer and model ?

daniele93dc, 2018-03-02:
Galaxy s5 and Asus Zenfone

austinkettner, 2018-03-03:
Can you also outline what version of CameraKit you are using here?

daniele93dc, 2018-03-03:
both 0.13.2 and 1.0.0

daniele93dc, 2018-06-17:
There are any news with new version for this?

emersoncloud, 2018-07-31:
@daniele93dc thanks for following up with this. It is not currently possible to create an instance from a service like the use case you mentioned. We will review this internally and add it to our milestones for future releases if we get enough interest! 

Our current focus has been delivering core camera functionality for 1.0.0!

closed 2018-07-31 by emersoncloud