CameraKit/camerakit-android #323: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.reconnect()

meMuhammadkamal, 2018-02-21:
The fragment started then closed very quickly this exception shows up:

    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.reconnect()' on a null object reference
                                                                        at com.wonderkiln.camerakit.Camera1.setupPreview(Camera1.java:696)
                                                                        at com.wonderkiln.camerakit.Camera1.start(Camera1.java:127)
                                                                        at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:266)
                                                                        at android.os.Handler.handleCallback(Handler.java:739)
                                                                        at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                        at android.os.Looper.loop(Looper.java:145)
                                                                        at android.os.HandlerThread.run(HandlerThread.java:61)

SergeyTrukhachev, 2018-03-07:
Same here. Google Pixel XL, Android Oreo.

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.reconnect()' on a null object reference
       at com.wonderkiln.camerakit.Camera1.setupPreview(Camera1.java:696)
       at com.wonderkiln.camerakit.Camera1.start(Camera1.java:127)
       at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:266)
       at android.os.Handler.handleCallback(Handler.java:790)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:164)
       at android.os.HandlerThread.run(HandlerThread.java:65)

austinkettner, 2018-03-08:
Should be resolved in #318 can you please test that out and I will reOpen this if you comment that it is still an issue.

closed 2018-03-08 by austinkettner

mptrista, 2018-06-01:
It is still an issue in 1.0.0-SNAPSHOT.