CameraKit/camerakit-android #324: Lock orientation to portrait - preview stretches

aravindhkumar23, 2018-02-22:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

com.wonderkiln:camerakit:0.11.2

### Steps to Reproduce

1. Open app in landscape mode -> preview stretches
2. Rotate to portrait mode -> again preview stretches

### Expected Behavior
Preview should not be stretched

### Actual Behavior
 Preview stretches on orientation change.
My app is locked to portrait mode and preview looks fine when launched in portrait mode, but if it is opened in landscape mode it stretches.

**Any suggestion to fix in this particular version. Or is it possible to lock the camera orientation to portrait mode?**

my xml code:
`<com.wonderkiln.camerakit.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:adjustViewBounds="true"
            app:ckZoom="pinch"
            app:ckFocus="tapWithMarker"
            app:ckPermissions="picture"
            android:background="@color/colorGray"
            />`

austinkettner, 2018-03-09:
This should be resolved in #318 - let me know if not and I can reOpen this.

closed 2018-03-09 by austinkettner

maruthivarman, 2020-05-30:
I face this error. how to set orientation issue with wonderkiln.camerakit.CameraView