CameraKit/camerakit-android #325: Selfie Screen flash not working - samsung s8+

aravindhkumar23, 2018-02-22:
### Is this a bug report?

Not sure

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

com.wonderkiln:camerakit:0.11.2

### Actual Behavior
 Selfie screen flash is not working, it works with normal flash(both front and back).
but some devices is provided with screen flash 

Tested in OnePlus-5, Samsung s8+

my xml code:
`<com.wonderkiln.camerakit.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:adjustViewBounds="true"
            app:ckZoom="pinch"
            app:ckFocus="tapWithMarker"
            app:ckPermissions="picture"
            android:background="@color/colorGray"
            />`

KRNareshKumar, 2018-04-18:
I am also facing a similar issue , where normal flash works fine but Screen flash is not working.

Could you provide with an update if there is any workarounds I can use to get over it.

Thanks in advance.

emersoncloud, 2018-07-31:
Can you test on our latest version 1.0.0-beta1 #318 and report back if you are still experiencing issues? I'll reopen this ticket if they still exist. Thanks! =)

closed 2018-07-31 by emersoncloud