CameraKit/camerakit-android #326: Camera Freezes

abhinav272, 2018-02-22:
### Is this a bug report?

No

I wonder why camera preview and app freezes when I call cameraview.captureVideo() and cameraview.stopVideo().

That is, whenever I start and end my video recording my app freezes, I can see frames skipped.
I am sure I am not doing anything blocking, Any Help?

austinkettner, 2018-02-22:
@abhinav272 Can you please provide device, CameraKit version, Android version, and other information on the issue? We will then reOpen

closed 2018-02-22 by austinkettner

abhinav272, 2018-02-22:
Android Version 7.1.1
Camera kit 0.0.13
Nexus 5

Do you require any other information?

austinkettner, 2018-02-22:
Nope, can you test https://github.com/CameraKit/camerakit-android/pull/318 this issue should be resolved in that. This was a noted issue in 0.13.x releases. Please comment in that PR with any issues, we will be quick to address.