CameraKit/camerakit-android #328: Video Recording does not return Video file on Pixel (Android 8.1)

AdrianMetzner, 2018-03-07:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes
### Environment

Google Pixel
Android 8.1.0
Lib version: 0.13.0 (also appears under 0.13.2)

### Steps to Reproduce
Part 1: 
1. Start Activity with camera view  which registers itself as the CameraKitEventListener at the view
2. Start Video capture via  captureVideo();
3. Stop Video capture after some time via stopVideo();

Part 2:
1. Start Activity with camera view  which registers itself as the CameraKitEventListener at the view
2. Start Video capture via  captureVideo(temp); with temp beeing an existing file
3. Stop Video capture after some time via stopVideo();

### Expected Behavior
Part1: onVideo(CameraKitVideo) should be called with an arbitrary VideoFile
Part2: on Video(CameraKitVideo) CameraKitVideo.getVideoFile() should return the recorded file instead of null!

### Actual Behavior
Part 1: onVideo(CameraKitVideo) is never called. CameraView Restarts (according to events)
Part 2: onVideo(CameraKitVideo) is called but CameraKitVideo.getVideoFile()  returns null
Note to Part 2: The video is recorded to the specified file. Just the CameraKitVideo.getVideoFile() value is null

TheHollidayInn, 2018-03-16:
This is happening on some devices with Android 7 as well. I'll post if I can get more info.

Barazu, 2018-05-28:
I've encountered the same problem (part 1), and granting external storage permission solved it.

MrDizzle, 2018-06-28:
I have granted WRITE_EXTERNAL_STORAGE and I'm still having this issue

emersoncloud, 2018-08-08:
Hi guys, I've added this to our 1.0.0 milestone. We will be adding video back shortly once we get the image capture API stabilized and I'll make sure this is addressed.