CameraKit/camerakit-android #331: Black camera preview as soon as 'Allow' is clicked (new permissions)

Zarkopafilis, 2018-03-08:
### Is this a bug report?
Yes
(Maybe)

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment
Nexus 6p API 25 on android studio emulator 

### Steps to Reproduce

Just add a camera view and set managing permissions 

(Write your steps here:)

1. Add camera view
2. Set managing Permissions to auto
3. Run on the emulator

### Expected Behavior

As soon as the user taps 'Allow' the camera preview should be displayed

### Actual Behavior

Preview turns black instead. If the activity/app is re-entered, the camera works properly though, but this is bad for user experience.

Zarkopafilis, 2018-03-11:
If the activity (or the fragment's activity) that holds the camera view listens for successful permission requests and then re-initializes the camera view, everything works properly, but this is a workaround.

emersoncloud, 2018-07-31:
Hey @Zarkopafilis can you test on the latest version 1.0.0-beta1? This issue should be resolved there. If not report back and I can reopen this ticket. =)

closed 2018-07-31 by emersoncloud

hiviiup, 2018-08-03:
Hey @emersoncloud ,I used 1.0.0-beta1.it's also black preview. 
[image]
[image]

emersoncloud, 2018-08-04:
Hi @hiviiup what steps did you take to get this blacked preview? We have another issue #348 to track this, should be resolved shortly but comment there if you have any more feedback.

barees63, 2019-05-09:
I'm having this exact issue also, after granting permission the (back) camera preview remains black, if I toggle between front & back camera it starts working.

Once the initial permissions have been granted the preview works correctly every time.

I have CameraKitView hosted in a fragment which is in a ViewPager.