CameraKit/camerakit-android #333: Select From Gallery Show All Folder

muratYamann, 2018-03-12:
**Hi CameraKit Team,**

How to show all photo files when selected galery.

[image]
[image]

emersoncloud, 2018-04-10:
Hi @muratYamann Could you fill out an issue following our issue template? I'm not sure what the expected behavior is and what is actually happening. Thanks!

emersoncloud, 2018-07-31:
Closing this issue. Let me know if you are still experiencing issues and I can reopen!

closed 2018-07-31 by emersoncloud