CameraKit/camerakit-android #334: Camera preview becomes severly stretched

rohanjain1, 2018-03-12:
I have implemented the camera kit by giving it matchParent bounds. 
I want it to cover the full screen. My app operates in portrait mode only.

Steps to recreate the issue : (100% reproducibility)
- launch the camera preview  (opens in portrait mode, aspect ratio is correct)
- rotate the device by 90 degrees so the device is in landscape orientation.
- Switch camera
- Notice that the preview gets extremely stretched.

Please let me know if there is a resolution i can implement on my end for it

austinkettner, 2018-03-12:
What version are you using?

closed 2018-03-12 by austinkettner

rohanjain1, 2018-03-12:
I'm using 0.13.1

austinkettner, 2018-03-12:
please try 1.0.0 out @rohanjain1

Gelassen, 2018-06-15:
Follow-up the conversation in the email. Experience stretched video in landscape mode - set up by default as well as after orientation change. 

Here is the sample to reproduce 
https://drive.google.com/open?id=1w42GlIiWCOqA31U8pDfbWK98uniH_zCA

Here is a initial description 
"Though I also noticed issue with stretching frame in landscape, there is a couple of threads in project issue tracker, one of them recommended  to move with 1.0.0 release. I tried it too, stil experience the issue, default launch in landscape even change movement vectors."