CameraKit/camerakit-android #338: Gradle Build error "Failed to resolve implementation 'com.camerakit:camerakit:1.0.0'"

jjpatel1996, 2018-03-20:
First of all sorry for removing your pre-written description. i was getting error when i add implementation 'com.camerakit:camerakit:1.0.0' in Gradle App Module. it giving error that "Failed to resolve implementation 'com.camerakit:camerakit:1.0.0'".

emersoncloud, 2018-03-21:
Hi @jjpatel1996 take a look at the Dependency section of the latest PR notes, #318. You should be using this implementation line `implementation group: 'com.camerakit', name:'camerakit', version:'1.0.0-Snapshot', changing: true`

jjpatel1996, 2018-03-21:
Thanks  @emersoncloud for response.. But it still giving me this error "Failed to resolve: com.camerkit:camerakit:1.0.0-Snapshot".

jjpatel1996, 2018-03-21:
Thanks @emersoncloud for help. It's working now.

closed 2018-03-21 by jjpatel1996