CameraKit/camerakit-android #341: Samsung S4 green screen in preview

chrbien, 2018-03-26:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment
Samsung S4

### Steps to Reproduce
Open a camera preview with CameraKit

### Expected Behavior

Camera preview shows surroundings.

### Actual Behavior

Preview shows a large green area, only part of surrounding is visible.

[image]

### Reproducible Demo
You should be able to reproduce using the sample app in the repo. 

This only happened for a customer of mine using a Samsung S4. All other test devices I used are not affected. But others seem to have the same (or similar) issues as described in #308

emersoncloud, 2018-03-26:
@chrbien thanks for the issue! Could you test on the latest PR #318? The green screen issue should be resolved, we haven't gotten any more reports of it being a problem. Let me know if that works.

chrbien, 2018-03-27:
Hello,
thanks for the fast response. I played around a bit with the current 1.0.0 version but it does not seem to be in a state that I can integrate easily into our current project. I do not own a S4 myself and I cannot give out a test version to my customer with so many important features still missing. But this is no time critical issue (at least for me). I will wait a bit and respond later after the project has made a bit more progress.

emersoncloud, 2018-03-27:
We are rewriting our documentation for 1.0.0, hopefully that will help with the integration on your end. I'll add this bug to the 1.0.0 release and I'll be sure to circle back when we've got the official release out! Thanks for your understanding.

android-dataticket, 2018-06-08:
@emersoncloud I dont think its documentation thats making it difficult.. I have an old forked version that's probably getting close to being a year old, that's been slightly tweaked for my application and I wanted to quickly try 1.0.0 without having to fork again and implement a folder library, I just wanted to implement quickly but it appears 1.0.0 is not available with jitpack..

emersoncloud, 2018-06-08:
Hi @android-dataticket. We've made pretty drastic changes to the code base over the past year. You're right we are not yet on jitpack but you can check the PR notes #318 to see how to implement v1.0.0.

closed 2018-07-25 by austinkettner