CameraKit/camerakit-android #345: Activate Flash like native camera when I Take a picture

bard0x, 2018-03-29:
Hi guys,
an information: when I invoke the method captureImage() with the status of the flash setted to FLASH_ON the flash of my back-camera is not activating. It's activated if I Tap the screen for one/two seconds only if the property of the ck is setted to "tapWithMarker".  ( app:ckFocus="tapWithMarker")
I would simulate the flash on pressing of the capture button like the native camera or like a comportment of the ckFocus="tapWithMarker" property.

Thank's in advance and thank's for this project guys!

ksc91u, 2018-05-16:
setMethod(CameraKit.Constants.METHOD_STANDARD) when flash light state is FLASH_ON

austinkettner, 2018-05-16:
Please take a look at our 1.0.0 PR! =) This along with the documented methods, are entirely ready to be used!

closed 2018-05-16 by austinkettner