CameraKit/camerakit-android #348: Blacked preview screen on many devices

CodeCombiner, 2018-04-05:
### Is this a bug report?

Many devices display camera preview screen blacked. I think it might be a problem with autoexposure. The device have partial support of Camera2 API and are not legacy completely.

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

(Write your answer here.)

### Environment

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

emersoncloud, 2018-04-10:
Hi @CodeCombiner thanks for using CameraKit. Could you fill out an issue template? That will help me get to the bottom of your problem. Thanks!

CodeCombiner, 2018-04-10:
Ok, I will install demo app and try to reproduce the issue.

epekel, 2018-04-12:
I am facing this issue as well. Testing on a Samsung Galaxy S6. Front camera seems to work fine, but back camera preview is a black screen. I toggle the facing with a button, and the method `cameraView.toggleFacing()`. When I pressed the button to `captureImage()`, I got this Exception message:

```04-12 16:42:55.690 19313-21184/com.formathletica.ble_android E/AndroidRuntime: FATAL EXCEPTION: CameraPreview@1523576572972
    Process: com.formathletica.ble_android, PID: 19313
    java.lang.NullPointerException: Attempt to read from field 'int com.camerakit.CameraKitView$Attributes.sensorOrientation' on a null object reference
        at com.camerakit.CameraKitView$CameraPreview$2.onImage(CameraKitView.java:1422)
        at com.camerakit.CameraKitView$Camera2$1$6.run(CameraKitView.java:2132)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)
```

Hopefully might narrow down the possible cause of this issue.

kheldiente, 2018-04-15:
Hi all, recently experienced this using 0.13.1. Rear camera preview is pure black. Changing it back to 0.13.0 fixed it. Hope this resolves your issue as well.

epekel, 2018-04-16:
@kheldiente Thanks for the suggestion. Forgot to mention that I am using [this](https://github.com/CameraKit/camerakit-android/pull/318) (release v1.0.0 snapshot)

GuoJinyu, 2018-04-27:
I find this too，as @CodeCombiner see，in some of devices， camera preview is much darker than the system camera app or other apps' camera function. Because of the exposure process?

austinkettner, 2018-04-27:
@epekel Can you provide some screenshots or examples? We will push fixes to the 1.0.0 release addressing this if we can trace it down. Device and OS version would be helpful too.

epekel, 2018-04-30:
@austinkettner 
Here is how I toggle front and back camera: 
```@OnClick(R.id.bttn_switch_camera)
    fun switchCamera() {
        mCameraView.toggleFacing()
        mCameraFacing = mCameraView.facing
    }```
(`mCameraFacing` is for persistence, not really important)

I capture the photo with the following line on capture button press: `mCameraView.captureImage(mCameraListener)`

The issue is with the back camera, the preview screen (i.e. the layout below) is completely black when switched to the back camera. The front camera works though. 

```<com.camerakit.CameraKitView
        android:id="@+id/camera_view"
        android:layout_width="match_parent"
        android:layout_height="0dp"
        android:layout_weight="9"
        android:adjustViewBounds="true"
        app:camera_imageJpegQuality="75"
        app:camera_focus="continuous"
        app:camera_permissions="camera"/>```

When trying to capture a photo in the back camera (with the preview being black), I get an NPE from the following code segment (decompiled .class, bold line is the point of exception):

```public void captureImage(final CameraKitView.CameraApi.ImageCallback callback) {
                    if (CameraKitView.this.mFlash == 1 && !this.mFlashing) {
                        try {
                            Camera2.this.mPreviewRequestBuilder.set(CaptureRequest.FLASH_MODE, 2);
                            Camera2.this.mCaptureSession.setRepeatingRequest(Camera2.this.mPreviewRequestBuilder.build(), (CaptureCallback)null, (Handler)null);
                            this.mFlashing = true;
                            (new Handler()).postDelayed(new Runnable() {
                                public void run() {
                                    captureImage(callback);
                                }
                            }, 1000L);
                            return;
                        } catch (Exception var3) {
                            ;
                        }
                    }

                    Camera2.this.background(new Runnable() {
                        public void run() {
                            CameraKitView.Size previewSize = Camera2.this.getAdjustedPreviewSize();
                            int width = Camera2.this.getWidth();
                            int height = Camera2.this.getHeight();
                            **float widthRatio = (float)width / (float)previewSize.getWidth();**
                            float heightRatio = (float)height / (float)previewSize.getHeight();
                            float ratio = Math.min(widthRatio, heightRatio);
                            if (widthRatio > 1.0F || heightRatio > 1.0F) {
                                ratio = Math.max(widthRatio, heightRatio);
                            }

                            Bitmap bitmap = Camera2.this.mTextureView.getBitmap((int)((float)previewSize.getWidth() * ratio), (int)((float)previewSize.getHeight() * ratio));
                            ByteArrayOutputStream stream = new ByteArrayOutputStream();
                            bitmap.compress(CompressFormat.JPEG, 100, stream);
                            byte[] byteArray = stream.toByteArray();
                            callback.onImage(byteArray);
                            if (mFlashing) {
                                try {
                                    Camera2.this.mPreviewRequestBuilder.set(CaptureRequest.FLASH_MODE, 0);
                                    Camera2.this.mCaptureSession.setRepeatingRequest(Camera2.this.mPreviewRequestBuilder.build(), (CaptureCallback)null, (Handler)null);
                                    mFlashing = false;
                                } catch (Exception var11) {
                                    ;
                                }
                            }

                        }
                    });
                }```

The error reports that `getWidth()` is performed on a null object reference, so previewSize is null (which makes sense, since there is no preview in the first place). So the error might be a bit misleading, but the issue is with Camera 2's (assuming this is the back camera) preview screen basically. Hope these help, not sure what else I could do to speed up the debugging process.

epekel, 2018-05-08:
Kind of a late update but just wanted to let you know that [this commit](https://github.com/CameraKit/camerakit-android/commit/78a7a8403c71101f8f7c6b30e845d7f3836678bc) seems to fix the issue, at least for the device I am testing with. (I posted the comment above after this commit so it seems it was not a device specific issue after all)

dhavalappspool, 2018-07-24:
+1 same issue

AndrewGable, 2018-07-24:
Seeing the same on `1.0.0-Beta1`
- Android OS Version: `7.1.2`
- Android Device: `LGE LM-X212(G)`

[image]

WhiteRaBot17, 2018-07-27:
Unable to reproduce bug on Galaxy S5.

However after reviewing the comments and going over commit history this PR is a potential candidate for the NPE:

https://github.com/CameraKit/camerakit-android/pull/280/files

There may be a condition not being caught when setting the previewSize. Will investigate further..

AndrewGable, 2018-08-01:
- Android version: `7.0`
- Phone model: `LG-Aristo`

------

- Model: `LGE LM-X210(G)`
- Version: `7.1.2`

------

- platformVersion - `7.1.2`
- deviceModel - `KYOCERA E6810`

-------

- Model: `LGE LG-TP260`
- Version: `7.0`

------

- Model: `LGE VS501`
- Version: `7.0`

AndrewGable, 2018-08-03:
-  Android Device: `E6810`
- Android OS version: `7.1.2`

-----

- Android Device: `LG-M210`
- Android OS version: `7.0`

------

-  Android Device: `LGMS210`
- Android OS version: `7.0`

------

- Android Device: `LM-X210(G)`
- Android OS version: `7.1.2`

------

- Android Device: `VS501`
- Android OS version: `7.0`

austinkettner, 2018-08-06:
Resolved in: 0f37519586446a56ce6d419646afc8ac32cca85f / #318 

Please comment if reproduced in Beta3 and I will reOpen

gsavvid, 2018-09-20:
Same here. Tested with beta3.9 on Nexus 5x with Android 8.0.1. No camera preview is shown, just a black screen.

emersoncloud, 2018-09-20:
Hi @gsavvid and @shreekrishnaban can you post the code you are using to start the CameraKitView?

gsavvid, 2018-09-25:
@emersoncloud here's a simple demo app with a screenshot. I've tested it on my Nexus 5x Android 8.1.0 (not 8.0.1 like I accidentally wrote previously).

emersoncloud, 2018-09-25:
On the latest beta, v1.0.0-beta3.9 you have to add two additional functions when creating a `CameraKitView`. There is a comment about this in the PR notes #318, but it's easy to miss. These methods won't stay for the final v1.0.0 version but were needed for this particular beta. 

In the `MainActivity.kt` you need to override `onStart()` and `onStop()` like so:

```
override fun onStart() {
    super.onStart()
    camera.onStart()
}

override fun onResume() {
    super.onResume()
    camera.onResume()
}

override fun onPause() {
    camera.onPause()
    super.onPause()
}

override fun onStop() {
    camera.onStop()
    super.onStop()
}
```

Thanks for your demo code, it was very helpful! We are working on updated documentation for the final 1.0.0 release, clearing up these implementation issues.

pikzelz, 2018-12-30:
I was also having the issue where the preview showed up black. Tried all solutions given here and nothing worked, preview stayed black.

After downloading the source code and adding it to my project I found out that resume() in CameraPreview.kt wasn't called and thus the preview didn't start. After digging a little further I noticed that the init call has a listener setup to wait for the surfaceView to become ready.

The original code:
```
cameraSurfaceView.cameraSurfaceTextureListener = object : CameraSurfaceTextureListener {
    override fun onSurfaceReady(cameraSurfaceTexture: CameraSurfaceTexture) {
        surfaceTexture = cameraSurfaceTexture
        surfaceState = SurfaceState.SURFACE_AVAILABLE
        if (lifecycleState == LifecycleState.RESUMED) {
            resume()
        }
    }
}
```

It appears that the lifecycleState isn't always RESUMED when onSurfaceReady is called. 
After changing it to:
```
cameraSurfaceView.cameraSurfaceTextureListener = object : CameraSurfaceTextureListener {
    override fun onSurfaceReady(cameraSurfaceTexture: CameraSurfaceTexture) {
        surfaceTexture = cameraSurfaceTexture
        surfaceState = SurfaceState.SURFACE_AVAILABLE
        if (lifecycleState == LifecycleState.STARTED || lifecycleState == LifecycleState.RESUMED) {
            resume()
        }
    }
}
```

The preview worked every single time.

JorgeDLS, 2019-03-12:
Having this problem with 1.0.0-beta3.11 with LG G4. Even tried with a 1.0.0-beta3.12 Snapshot with no luck.

xhiftcorp, 2019-03-20:
> Having this problem with 1.0.0-beta3.11 with LG G4 (Device model LG-H815). Even tried with a 1.0.0-beta3.12 Snapshot with no luck.

I had same problem with LG G4 and found it worked with 1.0.0-beta3.1

cunum, 2019-06-28:
> > Having this problem with 1.0.0-beta3.11 with LG G4 (Device model LG-H815). Even tried with a 1.0.0-beta3.12 Snapshot with no luck.
> 
> I had same problem with LG G4 and found it worked with 1.0.0-beta3.1

I observed the same 1.0.0-beta3.10 works for me, the newer milestones not. I suspect in the newer versions camera.onStart() and camera.onResume() conflict with each other.

wcmolina, 2019-11-30:
> I observed the same 1.0.0-beta3.10 works for me, the newer milestones not. I suspect in the newer versions camera.onStart() and camera.onResume() conflict with each other.

Same here, I'm using a Samsung Galaxy S8+ with Android 9. Preview seems to work fine if I use `1.0.0-beta3.10`, but I get a black preview if I use `1.0.0-beta3.11` instead.

caiocesarf, 2019-12-03:
> Same here, I'm using a Samsung Galaxy S8+ with Android 9. Preview seems to work fine if I use `1.0.0-beta3.10`, but I get a black preview if I use `1.0.0-beta3.11` instead.

fixed the black preview issue downgrading from v1.0.0-beta3.11 to v1.0.0-beta2. Using Samsung GT-192000 with Android 7.0

longld103, 2020-02-05:
In my case, I have fixed black preview by close another camera app. Black preview occur when camera opening by another application.

kabirsaheb, 2021-02-16:
I was conditionally adjusting camera view visibility, I also got blank screen. 

Calling onStop and onStart, when visibility is adjusted fixed it.
        cameraWrapper.visibility = View.VISIBLE
        camera.onStop()
        camera.onStart()

PR #474 (2019-01-23): v1.0.0-beta3.11
## CameraKit v1.0.0-beta3.11

**Issues covered in this release:**
Closes #348
Closes #426 
Closes #479