CameraKit/camerakit-android #351: Error preview in capture landscape

prestashopy, 2018-04-22:
### Is this a bug report?

Yes,  when I capture video in landscape, after the preview of the camera it is bad scaling.

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

I could solve it by adding the following lines in the SurfaceViewContainer.java file, layoutChild function:

    private void layoutChild(int width, int height) {
        final View child = getChildAt(0);

        int previewWidth = width;
        int previewHeight = height;
        if (mPreviewSize != null) {
            previewWidth = mPreviewSize.getWidth();
            previewHeight = mPreviewSize.getHeight();
        }

        /* PATCH */
        if(previewWidth > previewHeight){
            int tmp_previewWidth = previewWidth;
            previewWidth = previewHeight;
            previewHeight = tmp_previewWidth;
        }
        /* END */

        if (width * previewHeight > height * previewWidth) {
            final int scaledChildHeight = previewHeight * width / previewWidth;
            child.layout(0, (height - scaledChildHeight) / 2, width, (height + scaledChildHeight) / 2);
        } else {
            final int scaledChildWidth = previewWidth * height / previewHeight;
            child.layout((width - scaledChildWidth) / 2, 0, (width + scaledChildWidth) / 2, height);
        }
    }

### Environment

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

emersoncloud, 2018-04-27:
Hi @prestashopy thanks for bringing this to our attention. What version of Camera Kit are you using?

prestashopy, 2018-05-02:
Hi @emersoncloud,
I downloaded  "master"  version
      android: versionCode = "1"
     Android: versionName = "0.13.2"

In the end I used next code, and the preview scale works fine, it only produces error (sometimes, when you turn on and off) in the orientation and size (not full screen):

```
   private void layoutChild(int width, int height) {
        final View child = getChildAt(0);
        child.layout(0, 0, width, height);
 }
```

jatin-lakhani, 2018-05-04:
+1 
I'm still facing this issue.
Any solution will help a lot

Thanks in advance!

emersoncloud, 2018-05-04:
@prestashopy thanks for the information. We are moving to v1.0.0 very soon, its currently in review here: #318. 1.0.0 restructured and improved on a large portion of the 0.13.x codebase. Try out 1.0.0 and let me know if your issue persists.

closed 2018-05-18 by emersoncloud