CameraKit/camerakit-android #353: How can I use the image quality ? 100 still not very well

Qdafengzi, 2018-04-29:
The photo does not have a native camera with high detail, obvious noise, and graininess.

austinkettner, 2018-04-29:
Hey @Qdafengzi can you please provide more information on the methods used to capture, device, os, and reproduction steps? Also was this on CameraKit 0.13.x or 1.0.0-Snapshot?

Qdafengzi, 2018-05-15:
I'm sorry to have been busy working on the project, so late to answer this question, my project is for the factory environment, the camera parameters of the device is already relatively low, but compared with the camera's own camera photo screen is still somewhat blurred, preview The picture has yet to be optimized. Now there is another problem is how to refresh the preview interface after I finish the photo, if the picture is taken horizontally and the picture is taken vertically, the picture will be stretched and deformed. Is there a corresponding API that can be called?

Qdafengzi, 2018-05-15:
camerakit:0.13.1   Android 5.1.1   1920*1080   Qualcomm Technologies,Inc MSM8939

austinkettner, 2018-05-15:
v1.0.0 is slated to be fully released by the end of this month!

Qdafengzi, 2018-05-15:
I will deliver the product on the 18th. Is there a beat version?

austinkettner, 2018-05-15:
v1.0.0 is stable and we expect no changes to its API usage. We have heard nothing but positive feedback and improvements over previous generation 0.13.x and 0.12.x releases. Our suggestion, even in production, is v1.0.0. We have existing clients deploying with v1.0.0 already =)

Qdafengzi, 2018-05-15:
We have existing clients deploying with v1.0.0 already = ??

can  i use the v1.0.0 now ? Because I really need it now, I will actively submit bugs for you to test, thanks . can i ?

Qdafengzi, 2018-05-15:
Maybe I still don't use GitHub. I've seen the latest version branch. Thank you. 

I am too yong to sample , Laugh.

Qdafengzi, 2018-05-16:
in v1.0.0 , If I quickly click the camera button in the Demo, I haven’t finished taking pictures yet, I return to cause the program to crash. How do I release it?

`05-16 16:07:17.117 11977-12001/com.camerakit.app E/AndroidRuntime: FATAL EXCEPTION: CameraPreview@1526457794554
                                                                   Process: com.camerakit.app, PID: 11977
                                                                   java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.graphics.Bitmap.compress(android.graphics.Bitmap$CompressFormat, int, java.io.OutputStream)' on a null object reference
                                                                       at com.camerakit.CameraKitView$Camera2$1$7.run(CameraKitView.java:1855)
                                                                       at android.os.Handler.handleCallback(Handler.java:751)
                                                                       at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                       at android.os.Looper.loop(Looper.java:154)
                                                                       at android.os.HandlerThread.run(HandlerThread.java:61)`

austinkettner, 2018-05-16:
@emersoncloud Will log this in the PR notes and we will get it fixed in the demo app. He also may have thoughts on a release method. Expect a response soon!

Qdafengzi, 2018-05-21:
Regardless of the device, your Demo will be like this on any machine. Click and take a picture quickly, and then exit to appear. Release it. In addition, the zoom function has not been implemented yet?

emersoncloud, 2018-05-30:
@Qdafengzi Could you record your screen to show how you're causing the crash? I've tried to reproduce on several devices now. I don't know if we are taking the same steps to make the program crash. Thanks!

Qdafengzi, 2018-05-31:
@emersoncloud I will send Gif to your Gmail .  When will your v1.00 official release be released? Is there a specific time?

Qdafengzi, 2018-06-05:
@austinkettner Yes  The previous version had a zoom function. After using the 1.0 version, there was no Api call. I very much look forward to adding this feature to you. I will touch

Qdafengzi, 2018-06-14:
@austinkettner I'm sorry to interrupt you again, because the project I'm responsible for is handed over to the customer. In the middle of this month, the customer will accept the project. Because the zoom function of the integrated camera has not been implemented, it will cause our project to fail to be accepted successfully, so I have been asked you many times about the version of v1.0 with zoom function, I really need this function too much. I sincerely sorry for urging you many times. Thank you to all members of this open source project. Looking forward to release! Finally , can you tell me the exact time ? thank you . Because other members always tell me immediately and sure thing...

austinkettner, 2018-06-14:
We will aim to have this in by Monday for you.

emersoncloud, 2019-01-30:
Closing this issue. Please refer to ticket #447 for updates on zoom capabilities. It's currently set for the v1.0.0-beta4 milestone. Thanks!

closed 2019-01-30 by emersoncloud