CameraKit/camerakit-android #358: NullpointerException

ivanicsjozsef94, 2018-05-16:
### Is this a bug report?

yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

yes

### Environment
Android 7.0 Nokia

### Steps to Reproduce

(Write your steps here:)

1. Open camera.
2.  cameraView.captureImage(); is called.

(Write what happened. Add screenshots!)

### Reproducible Demo

takeImageButton.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View view) {
                if(cameraView != null) {
                    try {
                        cameraView.captureImage();
                    } catch (Exception e) {
                        Toast.makeText(PhotoBoothActivity.this, R.string.capture_image_error, Toast.LENGTH_LONG).show();
                    }
                }
            }
        });
[image]

emersoncloud, 2018-05-30:
Hi @ivanicsjozsef94 thanks for your issue. It looks like you're using CameraKit version 0.13.x. Very shortly we will officially move to v1.0.0, a much cleaner and more robust version of CameraKit. If you could give #318 a try and let me know if your issue persist that would be great!

emersoncloud, 2018-08-08:
Closing issue. I'll open if you still are having problems!

closed 2018-08-08 by emersoncloud