CameraKit/camerakit-android #360: Error inflating class com.wonderkiln.camerakit.CameraView

shaktisinh, 2018-05-21:
I have added following dependency in gradle files

`implementation 'com.wonderkiln:camerakit:0.13.1'`

 ```
repositories {
        maven { url "https://jitpack.io" }
    }
```

added this in xml file 
<com.wonderkiln.camerakit.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:adjustViewBounds="true"
        app:ckCropOutput="false"
        app:ckDoubleTapToToggleFacing="true"
        app:ckFacing="back"
        app:ckFlash="off"
        app:ckFocus="tapWithMarker"
        app:ckJpegQuality="100"
        app:ckMethod="standard"
        app:ckPinchToZoom="true"
        app:ckVideoQuality="highest"
        app:ckZoom="2.0" />

getting below error
`java.lang.RuntimeException: Unable to start activity ComponentInfo{com.howdoo/chat.hola.com.howdoo.camera.MainActivity}: android.view.InflateException: Binary XML file line #14: Binary XML file line #14: Error inflating class com.wonderkiln.camerakit.CameraView
    at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2416)
    at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476)
    at android.app.ActivityThread.-wrap11(ActivityThread.java)
    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
    at android.os.Handler.dispatchMessage(Handler.java:102)
    at android.os.Looper.loop(Looper.java:148)
    at android.app.ActivityThread.main(ActivityThread.java:5417)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
 Caused by: android.view.InflateException: Binary XML file line #14: Binary XML file line #14: Error inflating class com.wonderkiln.camerakit.CameraView
    at android.view.LayoutInflater.inflate(LayoutInflater.java:539)
    at android.view.LayoutInflater.inflate(LayoutInflater.java:423)
    at android.view.LayoutInflater.inflate(LayoutInflater.java:374)
    at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:287)
    at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139)
    at chat.hola.com.howdoo.camera.MainActivity.onCreate(MainActivity.java:47)
    at android.app.Activity.performCreate(Activity.java:6237)
    at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1107)
    at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2369)
    at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476) 
    at android.app.ActivityThread.-wrap11(ActivityThread.java) 
    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344) 
    at android.os.Handler.dispatchMessage(Handler.java:102) 
    at android.os.Looper.loop(Looper.java:148) 
    at android.app.ActivityThread.main(ActivityThread.java:5417) 
    at java.lang.reflect.Method.invoke(Native Method) 
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616) 
 Caused by: android.view.InflateException: Binary XML file line #14: Error inflating class com.wonderkiln.camerakit.CameraView
    at android.view.LayoutInflater.createView(LayoutInflater.java:645)
    at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:764)
    at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:704)
    at android.view.LayoutInflater.rInflate(LayoutInflater.java:835)
    at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:798)
    at android.view.LayoutInflater.inflate(LayoutInflater.java:515)
    at android.view.LayoutInflater.inflate(LayoutInflater.java:423) 
    at android.view.LayoutInflater.inflate(LayoutInflater.java:374) 
    at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:287) 
    at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139) 
    at chat.hola.com.howdoo.camera.MainActivity.onCreate(MainActivity.java:47) 
[... 40 lines cut ...]
    at com.wonderkiln.camerakit.CameraView.<init>(CameraView.java:99)
    at java.lang.reflect.Constructor.newInstance(Native Method) 
    at android.view.LayoutInflater.createView(LayoutInflater.java:619) 
    at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:764) 
    at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:704) 
    at android.view.LayoutInflater.rInflate(LayoutInflater.java:835) 
    at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:798) 
    at android.view.LayoutInflater.inflate(LayoutInflater.java:515) 
    at android.view.LayoutInflater.inflate(LayoutInflater.java:423) 
    at android.view.LayoutInflater.inflate(LayoutInflater.java:374) 
    at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:287) 
    at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139) 
    at chat.hola.com.howdoo.camera.MainActivity.onCreate(MainActivity.java:47) 
    at android.app.Activity.performCreate(Activity.java:6237) 
    at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1107) 
    at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2369) 
    at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476) 
    at android.app.ActivityThread.-wrap11(ActivityThread.java) 
    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344) 
    at android.os.Handler.dispatchMessage(Handler.java:102) 
    at android.os.Looper.loop(Looper.java:148) 
    at android.app.ActivityThread.main(ActivityThread.java:5417) 
    at java.lang.reflect.Method.invoke(Native Method) 
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616) `

emersoncloud, 2018-05-24:
Hello @shaktisinh. Could you link me to your project code so I can get a better idea of what might be causing this issue? Thanks!

shaktisinh, 2018-05-24:
It was conflation with other dependency. after renaming "surface_view_container" its working fine

closed 2018-05-24 by shaktisinh

3niman, 2018-06-04:
Hi ,

Where you have renammed this file of lib and with what name, as i can see there is one id of same. Please ans asap.

alsong, 2018-07-23:
I have got the same issue, I just forked the camera kit demo project and placed it on Android Studio and built it.

But I am not sure where am wrong

emersoncloud, 2018-07-28:
Hey @alsong take a look at the PR notes #318 and try implementing CameraKit v1.0.0.

yadavshivam3000, 2018-12-18:
Error inflating class com.wonderkiln.camerakit.CameraView
Caused by: java.lang.reflect.InvocationTargetException

Please help in resolving this.

aderinto05, 2019-03-13:
I can't find surface_view_container in my project or library.
any update for this issue? @shaktisinh