CameraKit/camerakit-android #364: Image get compressed twice?

nvmnghia, 2018-06-09:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Xiaomi Mi 5 MIUI 9.5.3.0

### Steps to Reproduce

There is a byte[] that seems to be an out-of-camera JPEG in CameraView@captureImage (it has EXIF, and of the same size as an images captured by the default camera). That byte[] is then passed to a PostProcessor, which outputs a JPEG that is almost 3 times larger than the initial byte[] on my device. Why does CameraKit recompress the image?

### Expected Behavior

Doesn't recompress the image.

(Write what you thought would happen.)

### Actual Behavior

CameraKit recompresses the image, making it 3 times larger (quality set to 100)
I saw no abnormal console output!

### Reproducible Demo

I'm just playing with the code. I admit that this could be a device-specific problem (I read somewhere that the DNG "path" of the Mi 5 is "broken", maybe the normal output is "broken" too). In that case, what is the byte[] that will be passed to PostProcessor, uncompressed Bitmap or even DNG? I want to at least know this piece of information :)
Thanks for your reply!

emersoncloud, 2018-07-02:
@nvmnghia thanks for the issue and I apologize for the late response. Which version of CameraKit are you using? The drive link doesn't seem to be working if you could post a new one that would be helpful. Thanks!

nvmnghia, 2018-07-03:
That was 0.13.1, the latest version in the README.md. I agree this could be the only solution as not all phone can change JPEG compression settings, but I think you should at least check the default settings (Mine is fixed at 85). 

I can't test the 1.0.0, as I can only capture preview frames, not real, full resolution picture! Is it a known bug, or you just haven't implemented it yet?

Thanks for your support, this piece of code does streamline my camera app!

nvmnghia, 2018-07-03:
Link was updated. My bad, it was `[gdrive bla bla](url)` =)))