CameraKit/camerakit-android #365: Native memory leak in METHOD_STILL mode

zakgof, 2018-06-10:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment
Seen on Android 4.2.2 and 6.0.1 phones
using com.wonderkiln:camerakit:0.13.2

### Steps to Reproduce

Taking pictures continuously in METHOD_STILL mode causes the native memory allocation to grow (ending up with OOM exceptions).

1. Setup `CameraView` in `METHOD_STILL` mode
2. Call `CameraView.captureImage`  multiple times continiously

### Expected Behavior
Memory allocation should remain nearly constant
### Actual Behavior
Native memory allocation grows until at some point OutOfMemoryError occurs.
This is seen in Android Studio Memory Profiler:

[image]

### Reproducible Demo

The complete activity class:
```
public class LeakingActivity extends AppCompatActivity {
    private ScheduledExecutorService scheduler;
    private CameraView cameraView;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        cameraView = findViewById(R.id.camera);
    }

    @Override
    protected void onResume() {
        super.onResume();
        if (ContextCompat.checkSelfPermission(this, Manifest.permission.CAMERA) != PackageManager.PERMISSION_GRANTED) {
            ActivityCompat.requestPermissions(this, new String[]{Manifest.permission.CAMERA}, 50);
        } else {
            startCam();
        }
    }

    @Override
    protected void onPause() {
        super.onPause();
        if (scheduler != null) {
            scheduler.shutdown();
            cameraView.stop();
        }
    }

    @Override
    public void onRequestPermissionsResult(int requestCode, String permissions[], int[] grantResults) {
        if (requestCode == 50 && grantResults.length > 0 && grantResults[0] == PackageManager.PERMISSION_GRANTED) {
            startCam();
        }
    }

    private void startCam() {
        cameraView.start();
        cameraView.setMethod(CameraKit.Constants.METHOD_STILL); // IMPORTANT !
        scheduler = Executors.newScheduledThreadPool(1, runnable -> new Thread(runnable,"Camera"));
        scheduler.submit(this::tick);
    }

    private void tick() {
        System.out.println("Capturing image...");
        cameraView.captureImage(cameraKitImage -> {
            System.out.println("Image captured bytes=" + cameraKitImage.getJpeg().length);
            scheduler.schedule(this::tick, 100, TimeUnit.MILLISECONDS);
        });
    }

}
```
activity_main.xml contains
`<com.wonderkiln.camerakit.CameraView android:id="@+id/camera" android:layout_width="240dp" android:layout_height="wrap_content" android:adjustViewBounds="true" android:keepScreenOn="true" />
`

emersoncloud, 2018-07-02:
@zakgof Thanks for the detailed report. We are going to move shortly to the official release of 1.0.0 which encompasses stability improvements and will fix issues like you're seeing here. There is currently a stable beta of 1.0.0 in PR #318. After the official release our attention will be on support and development of 1.0.0. 

I would encourage you to implement your application on 1.0.0 and see if the problem persists. In the mean time I will look into your issue here.

frontiertsymbal, 2018-07-29:
Have you date of release 1.0.0 version? I have the same issue.

austinkettner, 2018-07-29:
@frontiertsymbal you can use it right now via #318, we will have an update soon on the official release date.

emersoncloud, 2018-08-08:
@zakgof any updates from testing on the 1.0.0 branch? The current release is beta3.1, give it a shot!

zakgof, 2018-08-09:
Looks like setMethod is not yet implemented, so I cannot test:

[image]

prakaz, 2019-01-28:
will setMethod() be implemented?