CameraKit/camerakit-android #367: captureVideo   camera.setPinchToZoom(true);   RuntimeException

ZhangTengCR7, 2018-06-15:
java.lang.RuntimeException: setParameters failed
                                                                            at android.hardware.Camera.native_setParameters(Native Method)
                                                                            at android.hardware.Camera.setParameters(Camera.java:2011)
                                                                            at com.wonderkiln.camerakit.Camera1.setZoom(Camera1.java:295)
                                                                            at com.wonderkiln.camerakit.CameraView.setZoom(CameraView.java:375)

emersoncloud, 2018-07-02:
Hey @ZhangTengCR7 which version of CameraKit are you using? It would be helpful for the team and I if you could comment following our issue template so we can better diagnose your issue. Thanks!

emersoncloud, 2018-07-18:
Thanks for getting back to me @ZhangTengCR7, can you try to reproduce this issue on v1.0.0 beta? The beta is currently stable and will be the focus of CameraKit moving forward. This issue should already be fixed there! 

Check out the PR notes for implementation details to get started. #318. I'll reopen this issue if your problem persists. Thanks!

closed 2018-07-18 by emersoncloud