CameraKit/camerakit-android #369: setParameters failed

platinum7919, 2018-06-19:
Got the follow exception when starting the camera on Galaxy s6
`
06-19 10:37:15.328 2936-2936/com.charaku.popsical.tv D/RecorderActivity: onEvent:  CameraKitError
06-19 10:37:15.328 2936-2936/com.charaku.popsical.tv E/RecorderActivity: setParameters failed
    java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:1940)
        at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:804)
        at com.wonderkiln.camerakit.Camera1.adjustCameraParameters(Camera1.java:779)
        at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:671)
        at com.wonderkiln.camerakit.Camera1.start(Camera1.java:124)
        at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:266)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)
`

platinum7919, 2018-06-25:
I was playing around with the source and noticed that the crash stops after I disable the invertPreviewSizes inside Camer1.getPreviewResolution

```
  //boolean invertPreviewSizes = (this.mCameraInfo.orientation + this.mDeviceOrientation) % 180 == 90;
  boolean invertPreviewSizes = false;
```

The problem is that the preview will then be stretched in portrait mode (front camera). Perhaps this means that the size from getSupportedPreviewSizes cannot just be inverted manually like that, maybe the surfaceview needs to do some cropping manually?

emersoncloud, 2018-07-02:
@platinum7919 Thanks for the issue and for looking into this. We have been working hard on CameraKit version 1.0.0 which has many stability and use-ability improvements over 0.13.x. It is currently in a stable Beta, take a look at the notes here #318. 

Could you test your implementation on 1.0.0 and let me know if the problem persists? Thank you!

koala99, 2018-07-05:
I used 0.13.x, some mobile record video error , I find the error messsages 'setParameters failed'.
so i can't record .

musicode, 2019-03-08:
0.13.5 +1

anathefish, 2019-05-31:
0.13.4 +1, 'setParameters failed' on record video attempt

omid-io, 2020-01-22:
+1 on 0.13.5. Just seeing the "setParameters failed" msg wrapped in a RuntimeException logged under the CameraKitEventListener#onError(CameraKitError) callback.