CameraKit/camerakit-android #370: CameraKitView.VideoCallback never fires after CameraKitView.stopVideo is called

MrDizzle, 2018-06-19:
When using this setup up:

```
CameraKitView.VideoCallback videoCallback = (cameraKitView, videoObject) -> 
{
       onVideoCapture.onVideoCapture(videoObject);
};
cameraV.captureVideo(videoCallback);
cameraV.startVideo();
```

Calling

   `cameraKitView.stopVideo`

Does not fire videoCallback listener

MrDizzle, 2018-06-26:
WRITE_EXTERNAL_STORAGE is already enabled

MrDizzle, 2018-06-28:
This is for version 1.0.0

MMohamadPoor, 2018-07-25:
I have the same problem, how to get captured video in version 1.0.0 ? The video callback listener never fire...

austinkettner, 2018-07-25:
@MMohamadPoor Currently 1.0.0 Beta1 and Beta2 do not have Video support. We expect to have this in shortly, but has been a big challenge to get right.

ZhangTengCR7, 2018-07-30:
@MMohamadPoor Currently 1.0.0 Beta1 and Beta2 do not have Video support. We expect to have this in shortly, but has been a big challenge to get right.  emmmm.......

MrDizzle, 2018-08-11:
I'm using beta 3.1 with the added onStart/onStop methods, but it appears the video capture functionality is still not there, can we expect this in beta4 or has the usage changed as this callback still isn't firing for me?

austinkettner, 2018-08-11:
@MrDizzle Video is not in yet. We expect this either in Beta4 or Beta5. We will update the PR on release of Beta4 to outline what's new and if video is included.

MrDizzle, 2018-08-11:
Thanks for the update!

HBiSoft, 2019-05-21:
@austinkettner Any update on this?