CameraKit/camerakit-android #374: Updated Documentation for v1.0.0

emersoncloud, 2018-07-18:
Create a new set of documentation for the upcoming 1.0.0 release. 

Official Maintainer Ticket: @emersoncloud

Daksh14, 2018-07-21:
We are kinda clashing with this http://docs.camerakit.website/#/ and this [#318](https://github.com/CameraKit/camerakit-android/pull/318)

austinkettner, 2018-07-21:
@Daksh14 Right now the only documentation valid is the content in the PR Notes. We are working to update the docs site to be fully reflective of 1.0.0 by Release

Daksh14, 2018-07-21:
@austinkettner can i email you some errors where you can help?

austinkettner, 2018-07-21:
@Daksh14 Yes, you can use austin at wonderkiln.com

pinigtech, 2018-08-07:
Can I volunteer to help with the documentation?

emersoncloud, 2018-08-08:
Hey @pinigtech thanks for the offer! I'll have you email austin at camerakit dot email and he can cover this in more detail.

pinigtech, 2018-08-08:
Will do

emersoncloud, 2019-01-30:
We have updated the documentation for each beta version on our new docs page, https://camerakit.io/docs. Let me know if you have any more concerns.

closed 2019-01-30 by emersoncloud