CameraKit/camerakit-android #377: How to zoom the view with 1.0.0-beta1

hiviiup, 2018-07-23:
Hi,I have two question to ask:
1. how to zoom the view ?
I used  cameraView.setZoomFactor(Float),It is invalid.

2. I want to use it to implement face detection. But ,I cant find the method or listener  to get the camera data flow?

emersoncloud, 2018-08-08:
@hiviiup zoom has not yet been implemented in 1.0.0 but it will be before the official release. As for face detection and camera data flow, that is a feature we are looking to add in a later release. It is similar to #343. I'll add a note there with your use case in mind.

noamshio, 2018-08-28:
So as of now, there is no option to zoom the camera ? not even using pinch ?

CabreraGerardo, 2019-10-15:
Still nothing?

goyourfly, 2020-05-21:
Still nothinig?

tian102, 2020-10-14:
Alrighty guys... Any news regarding Zoom functionality? -2020