CameraKit/camerakit-android #379: CameraView not close on .close();

Trusted97, 2018-07-24:
### Is this a bug report?
Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?
Yes

### Environment
Camera Kit version : 0.13.1
Android 7.0 Api 24 (Samsung Galaxy S7 (SM-G930F) 

### Steps to Reproduce

1. Create an Activity with a CameraView and another activity without it
2. On second activity pass a int value in intent in first activiy get value with getIntent and if val > 0 close/release the camera

### Expected Behavior

The camera should be released 

### Actual Behavior

The value is correctly passed, because other code inside the if  statement work well but 
`CameraView.close();` doesn't work  the camera is still open

### Reproducible Demo

w.i.p

austinkettner, 2018-07-26:
Hey @Trusted97 give this a test on 1.0.0 via #318, let me know if this issue still exists!

closed 2018-07-26 by austinkettner

Trusted97, 2018-07-30:
Hi @austinkettner where can i found Constants in 1.0.0 version?

austinkettner, 2018-07-30:
Can you elaborate on what you mean @Trusted97 ?

Trusted97, 2018-08-03:
The constants about Method,Flash,VideoQuality...
`CameraKit.Constants.METHOD_STILL;`