CameraKit/camerakit-android #382: Camera Preview doesn't fit in Dialog

King-Henry, 2018-07-27:
Right now our team is undergoing an effort to support tablet views in our app, which involves using a camera inside a dialog. As part of this effort we're working at trying out different CameraKit versions to see what works best (0.13.2, Beta1, Beta2). We're having different issues in each.

In Beta1 the camera preview fits in the dialog just fine on my API 21+ device, but doesn't rotate. On Beta2 the camera preview rotates, but it no longer fits in the dialog.

It seems that Beta1 is making use of Camera2 on API21+ and the TextureView plays nicely with dialogs, but in Beta2 it seems stuck on Camera1 in which SurfaceViews don't play well with dialogs.

Any reason Beta2 stopped using Camera2? Are there intentions to go back to Camera2 in future versions? Any recommendations?

### Steps to Reproduce

1. Create a Theme that inherits `Theme.AppCompat.Light.Dialog` or `Theme.AppCompat.Light.DialogWhenLarge`

2. Apply theme to activity that contains the CameraKitView

### Expected Behavior

Camera Preview fits inside the constraints of the dialog window.

### Actual Behavior

Camera Preview expands outside the bounds of the dialog window.

austinkettner, 2018-07-28:
Hey @King-Henry can you email me at Austin [at] camerakit.email?

austinkettner, 2018-08-28:
Resolved in Beta4 #403

closed 2018-08-28 by austinkettner