CameraKit/camerakit-android #383: Camera not focussing, pictures are blurry

AndrewGable, 2018-07-27:
### Is this a bug report?

Yes!

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes!

### Environment
- Android Version: 6.0.1
- Device: Samsung note 4

---- 

- Android Version: 6.0.1
- Device: Samsung SM-G900V

---- 

- Android Version: 6.0
- Device: SM-G900V

----

- Android Version: 6.0.1
- Device: SM-T813

### Steps to Reproduce

Attempt to take a photo, the photo appears blurry.

### Expected Behavior

I would expect the camera auto focus and not be blurry

austinkettner, 2018-07-28:
Received and investigating @AndrewGable

AndrewGable, 2018-07-30:
More devices if it's helpful:

- Android Version: `6.0.1`
- Android Model: `samsung SM-G860P`

-----

- Android Device: `SM-N910V`
- Android OS version : `6.0.1`

AndrewGable, 2018-08-01:
- Android Device — `SM-N910V`
- Android OS version — `6.0.1`

-----

- platformVersion- `6.0.1`
- deviceModel- `samsung SM-G860P`

------

- Device: `SAMSUNG-SM-N915A`
- Android OS: `6.0.1`

------

- Model: `samsung SM-N910V`
- Version: `6.0.1`

---------

- Platform Version: `6.0.1`
- Model: `samsung SAMSUNG-SM-G`

--------

- platformVersion: `6.0.1`
- deviceModel: `samsung SM-G900V`

AndrewGable, 2018-08-01:
Oops, @austinkettner - I messed up the assignees and I don't think I can assign back @WhiteRaBot17, sorry!

austinkettner, 2018-08-01:
All corrected =)

capezzbr, 2018-08-03:
- Device: `samsung SM-G900F`
- Android OS: `6.0.1`

austinkettner, 2018-08-06:
Should be improved if not resolved in: 0f37519586446a56ce6d419646afc8ac32cca85f

austinkettner, 2018-08-06:
Resolved in #318

austinkettner, 2018-08-28:
Reported on 

Device: `Samsung J5`
Android OS: `6.0.1`

sandys, 2018-09-22:
hi guys,
quick question - is there a way we can detect bad quality/blurry pictures ? we are facing this issue with camerakit and while we wait for a fix... im wondering if there is a way we can detect bad quality pictures so that we can ask the user to retake the picture

emersoncloud, 2018-11-19:
@sandys Can you comment with the Android Device, Android Version, and CameraKit Version you've seen the blurry picture issue on?

sandys, 2018-11-19:
Hi,
So we removed the camerakit version from our apps because of this issue. Just to be clear - we don't think it's a bug on CameraKit, just that we would like a way to manage it.

The phone were lots of Micromax, Xolo phones in India - which are lower end phones. The Android version is 6.0 (from what I last remember).

To reiterate - we are not looking for you to fix blurry images. If you can just detect it, we can give this feedback to customers and ask them to click again.

austinkettner, 2018-11-19:
@sandys we are working to improve results across devices in Beta4, but we are also looking at ways to 'detect anomalies' as you requested. I don't have a direct response at the moment in terms of when, but beta4 will improve the situation and we will see what we can do in that release regarding this.

sandys, 2018-11-19:
Thanks for your reply and for working on this.

I hope I'm not annoying, but want to reiterate the importance of anomaly detection versus performance improvement - users are happy to click the photo even ten times if you tell them what's wrong.
That will instantly help tons of users in a very impactful way.

Zhuxf1111, 2018-11-30:
Device : HUAWEI Honor 6X 
Device ID : BLN-AL40
This device also can't capture image.

emersoncloud, 2018-12-03:
@Zhuxf1111 which version of CameraKit are you using? Can you elaborate a bit more on what you mean by the device can't capture an image?