CameraKit/camerakit-android #385: Video is rotated, preview is distorted, when recording in a landscape-locked app

DLIMedia, 2018-07-30:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

Tested on Samsung Galaxy S7 Edge as well as emulation.

App that CameraKit is being used in is exclusively in LANDSCAPE, and locked in to that state through the AndroidManifest.xml file.

Using CameraKit external library, v0.13.2. No modifications have been made to the library by us.

### Steps to Reproduce

App Info: Single activity, CameraView control spans the entire view.

Issue 1:
1. Phone is held in landscape and NOT rotated
2. Attempt to record a video

Issue 2:
1. Phone is held in portrait, even though the app is in landscape, and NOT rotated
2. When the preview appears, swap out of the application and swap back to the app

This bug has seemingly reported in the past, mixed in with the distorted view that one could get (see below). Despite being reportedly fixed in several versions (12.x, I believe), it still has issues.

### Expected Behavior

Video records normally with the end product being in the correct orientation. Preview should not be distorted.

### Actual Behavior

Issue 1:
The video records successfully, but it is rotated 90 degrees counter-clockwise.

Issue 2:
The video technically records normally in that it records the portrait view, but this is wrong since the application is in landscape and the end result should be rotated.

Furthermore, when you swap out of the app and back in, without rotating the device, the camera preview is wildly distorted.

### Reproducible Demo

For various reasons - some of them legal - I cannot provide a reproducible demo.

emersoncloud, 2018-07-30:
Hi @DLIMedia thanks for the detailed issue. We are currently gearing up for the 1.0.0 release which will include video support to fix the issues you've mentioned above. I'll dig into your issue here in the mean time to see if it's an easy fix. However the best thing to do is to wait for that new 1.0.0 release! =)

austinkettner, 2018-07-30:
Also @DLIMedia can you shoot me an email to Austin [at] camerakit.email? Just reference this ticket, and I will send over some questions, we want to work more closely with companies using CameraKit so we can tailor critical updates as needed.

DLIMedia, 2018-07-30:
My concern is that I'm on a reasonably strict timeline and, of all the camera libraries I've used, this is the best one I've seen by far for what I want to do. 

And I've looked in to 1.0.0... that's somewhat of a big jump from 0.13.2 and I couldn't get it working right out of the box, so I'm hesitant. So, if possible, I'd appreciate a 0.13.2 fix... or at least the necessary guidance so that I can attempt to fix it myself.

Keep up the awesome work, though! :)

austinkettner, 2018-08-01:
@DLIMedia Shot you an email, check it out, ideally we can get you into Slack and troubleshoot the implementation of 1.0.0 as that should resolve most of your issues. We had a client update from 0.13.x with minimal code changes.