CameraKit/camerakit-android #389: Taking picture immediately after onResume crash

emersoncloud, 2018-08-01:
As reported by Ben in Slack: 

Tested on device.  results:
* If I call onResume captureImage immediately, 100% crash rate.
* If I call onResume delay 3 seconds captureImage - crashes on my file write permission denied.  (But is able to capture the image)
* if I uninstall the app completely, then the "ok to allow camera permissions" popup takes more than the 3 second delay, resulting in a crash.
* so I'm thinking race condition.

```
override fun onResume() {
       super.onResume()
       Log.d(TAG, "onResume")
       cameraKitView.onResume()
       cameraKitView.captureImage { _, imageBytes -> Log.i(TAG, "onImage ${imageBytes.size}") }
       Log.d(TAG, "captureImage")
  }
```

hiviiup, 2018-08-03:
I think  you can call the method on PreviewListener
        
cameraKitView.setPreviewListener(new CameraKitView.PreviewListener() {
            @Override
            public void onStart() {
                    //can take photo
            }

            @Override
            public void onStop() {

            }
        });

salamanders, 2018-08-03:
Makes sense.  Should captureImage secretly block until  PreviewListener onStart finishes to avoid crashes?

emersoncloud, 2018-11-19:
Resolved in the latest beta version, beta3.9. See implementation instructions [here](https://github.com/CameraKit/camerakit-android). If you still run into issues comment back here and I'll reopen this ticket!

closed 2018-11-19 by emersoncloud