CameraKit/camerakit-android #391: Default ckView.captureImage seems to take less than full-res photo

salamanders, 2018-08-01:
### Is this a bug report?
Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?
No.

### Environment
Pixel 2, Android P latest public beta

### Steps to Reproduce
Try to do the minimal "launch app, snap picture, write to file" hello world using the help that the slack folks provided (thank you!)

Exact code https://gist.github.com/salamanders/0e20fe1122000220c76b118d16c83f56
But the only interesting part is here https://gist.github.com/salamanders/0e20fe1122000220c76b118d16c83f56#file-mainactivity-kt-L32

1. App launches
2. App takes picture.
3. App Saves picture

### Expected Behavior
4032x3024 JPEG

### Actual Behavior
1080x1440 JPEG (According to photos app)

### Reproducible Demo
https://gist.github.com/salamanders/0e20fe1122000220c76b118d16c83f56#file-mainactivity-kt-L32

salamanders, 2018-08-01:
Ok, now I have read the guidelines.

emersoncloud, 2018-10-16:
Hi @salamanders in the latest beta, 1.0.0-beta3.9 you can set the image resolution through the function `cameraKitView.setImageMegaPixels(float megaPixels)`. We will be updating our documentation to cover this and more with beta4!

closed 2018-10-16 by emersoncloud