CameraKit/camerakit-android #392: An error occurred while connecting to camera 0: Status(-8): '7: connectHelper:1338: Higher-priority client using camera, ID "0" currently unavailable'

tommienu, 2018-08-03:
Ran into the following issue while developing an app and using my Pixel2XL and device. 

A reboot solved the issue and i'm certain that i caused this while building back and forth, but it would be nice to be able to catch this error and silently suppress it if it ever did happen to a user. A try/catch did not work for me.

Thank   ##you.

```
W/CameraBase: An error occurred while connecting to camera 0: Status(-8): '7: connectHelper:1338: Higher-priority client using camera, ID "0" currently unavailable'
V/FA: Connection attempt already in progress
E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
    Process: com.tompod.wearnotes, PID: 7006
    java.lang.RuntimeException: Fail to connect to camera service
        at android.hardware.Camera.<init>(Camera.java:519)
        at android.hardware.Camera.open(Camera.java:365)
        at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:667)
        at com.wonderkiln.camerakit.Camera1.start(Camera1.java:124)
        at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:266)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```

austinkettner, 2018-08-04:
Hi @tommienu what version of CameraKit are you using?

tommienu, 2018-08-06:
Hi @austinkettner,

0.13.1

Thanks for a great library.

austinkettner, 2018-08-06:
You are welcome @tommienu can you try out 0.13.x or our v1.0.0-beta3 release and report back?

emersoncloud, 2018-10-15:
Closing this ticket. If it is still an issue leave a comment and I can re-open.

closed 2018-10-15 by emersoncloud