CameraKit/camerakit-android #393: Captured image has always the same ratio

antoineos, 2018-08-03:
### Is this a bug report?
Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?
No

### Environment
S8 : Oreo
Xperia L2 : Oreo
S5 mini : KitKat

### Steps to Reproduce

1. Use cameraKitView with camera_aspectRatio or custom height
2. Call captureImage
3. Display the photo in imageview

### Expected Behavior

The result of captureImage and the preview should be the same.

### Actual Behavior

Whatever the preview look like, the captured image ratio has always the same.

On the S8 even if my preview is a square or anything else, i will always get an image with a 1.33 ratio.

austinkettner, 2018-08-04:
Which version of CameraKit are you using @antoineos ?

antoineos, 2018-08-04:
Hi @austinkettner ,
1.0.0-beta1

pinigtech, 2018-08-06:
I am testing this on One Plus 6. I am not seeing an option to set camera_aspectRatio. I am using com.wonderkiln:camerakit:0.13.2. Do I need to upgrade to get that option?

[image]

antoineos, 2018-08-06:
@pinigtech 
yes : https://github.com/CameraKit/camerakit-android/pull/318

emersoncloud, 2018-08-06:
@antoineos @pinigtech We have just release 1.0.0-beta3 here: 0f37519 / #318 be sure to test on beta3 rather than beta1!

pinigtech, 2018-08-07:
With 1.0.0 branch I was getting no feedback (in ImageCallback) to tell me
whether the picture was portrait or landscape. The height and width was the
same in both cases. This was not the issue with the default branch thus was
using it.
Let me try the beta3 and update

antoineos, 2018-08-16:
@emersoncloud @austinkettner I just test with 1.0.0-veta3.1 and still have the same issue.
Here is my code : https://github.com/antoineos/CameraKitBetaTest
Again the preview is as i want but the picture i get is always 1920 * 1080

Am i doing something wrong ? Don't you have the same issue ?

emersoncloud, 2018-08-19:
hi @antoineos right now you have some control over the image resolution with the `cameraKitView.setImageMegaPixels();` function but that implementation will likely change and be refined in upcoming beta releases.

Username5263, 2018-10-05:
Hi! Has this been resolved by beta4? And if it is, is it available to utilize? Thanks!

emersoncloud, 2018-10-16:
Hi @Username5263 in beta3.9 you can set the aspect ratio with: `cameraKitView.setAspectRatio(float ratio)`. It takes a single float that is the ratio between height and width.

closed 2018-10-16 by emersoncloud

AudaciousSam, 2019-07-04:
I've tried to set aspect ratio, but is getting the same problem. 

Has it been solved since this post?