CameraKit/camerakit-android #394: CameraPreview Crash

austinkettner, 2018-08-04:
```

#0. Crashed: CameraPreview@1533401855192
       at com.camerakit.CameraKitView$Camera2$1$3.run(CameraKitView.java:1737)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:158)
       at android.os.HandlerThread.run(HandlerThread.java:61)

--

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference
       at com.camerakit.CameraKitView$Camera2$1$3.run(CameraKitView.java:1737)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:158)
       at android.os.HandlerThread.run(HandlerThread.java:61)

#0. Crashed: CameraPreview@1533401855192
       at com.camerakit.CameraKitView$Camera2$1$3.run(CameraKitView.java:1737)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:158)
       at android.os.HandlerThread.run(HandlerThread.java:61)

#1. OkHttp ConnectionPool
       at java.lang.Object.wait(Object.java)
       at io.intercom.okhttp3.ConnectionPool$1.run(ConnectionPool.java:67)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

#2. Crashlytics Exception Handler1
       at dalvik.system.VMStack.getThreadStackTrace(VMStack.java)
       at java.lang.Thread.getStackTrace(Thread.java:580)
       at java.lang.Thread.getAllStackTraces(Thread.java:522)
       at com.crashlytics.android.core.CrashlyticsUncaughtExceptionHandler.writeSessionEvent(CrashlyticsUncaughtExceptionHandler.java:991)
       at com.crashlytics.android.core.CrashlyticsUncaughtExceptionHandler.writeFatal(CrashlyticsUncaughtExceptionHandler.java:395)
       at com.crashlytics.android.core.CrashlyticsUncaughtExceptionHandler.handleUncaughtException(CrashlyticsUncaughtExceptionHandler.java:298)
       at com.crashlytics.android.core.CrashlyticsUncaughtExceptionHandler.access$200(CrashlyticsUncaughtExceptionHandler.java:56)
       at com.crashlytics.android.core.CrashlyticsUncaughtExceptionHandler$7.call(CrashlyticsUncaughtExceptionHandler.java:278)
       at com.crashlytics.android.core.CrashlyticsUncaughtExceptionHandler$7.call(CrashlyticsUncaughtExceptionHandler.java:275)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at io.fabric.sdk.android.services.common.ExecutorUtils$1$1.onRun(ExecutorUtils.java:75)
       at io.fabric.sdk.android.services.common.BackgroundPriorityRunnable.run(BackgroundPriorityRunnable.java:30)
       at java.lang.Thread.run(Thread.java:818)

#3. FinalizerWatchdogDaemon
       at java.lang.Thread.sleep(Thread.java)
       at java.lang.Thread.sleep(Thread.java:1031)
       at java.lang.Thread.sleep(Thread.java:985)
       at java.lang.Daemons$FinalizerWatchdogDaemon.sleepFor(Daemons.java:288)
       at java.lang.Daemons$FinalizerWatchdogDaemon.waitForFinalization(Daemons.java:299)
       at java.lang.Daemons$FinalizerWatchdogDaemon.run(Daemons.java:247)
       at java.lang.Thread.run(Thread.java:818)

#4. IntercomNexus-2
       at java.lang.Object.wait(Object.java)
       at java.lang.Thread.parkFor$(Thread.java:1220)
       at sun.misc.Unsafe.park(Unsafe.java:299)
       at java.util.concurrent.locks.LockSupport.park(LockSupport.java:158)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2013)
       at java.util.concurrent.ScheduledThreadPoolExecutor$DelayedWorkQueue.take(ScheduledThreadPoolExecutor.java:1085)
       at java.util.concurrent.ScheduledThreadPoolExecutor$DelayedWorkQueue.take(ScheduledThreadPoolExecutor.java:806)
       at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1036)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1098)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
[... 678 lines cut ...]

#61. pool-21-thread-1
       at java.lang.Object.wait(Object.java)
       at java.lang.Thread.parkFor$(Thread.java:1220)
       at sun.misc.Unsafe.park(Unsafe.java:299)
       at java.util.concurrent.locks.LockSupport.park(LockSupport.java:158)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2013)
       at java.util.concurrent.LinkedBlockingQueue.take(LinkedBlockingQueue.java:410)
       at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1036)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1098)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

#62. pool-6-thread-1
       at java.lang.Object.wait(Object.java)
       at java.lang.Thread.parkFor$(Thread.java:1220)
       at sun.misc.Unsafe.park(Unsafe.java:299)
       at java.util.concurrent.locks.LockSupport.park(LockSupport.java:158)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2013)
       at java.util.concurrent.LinkedBlockingQueue.take(LinkedBlockingQueue.java:410)
       at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1036)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1098)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
```

pinigtech, 2018-08-06:
I am also facing this with **com.camerakit:camerakit:1.0.0-beta3** On One Plus 6 phone. Get the following error:

java.lang.NoClassDefFoundError: Failed resolution of: Lkotlin/jvm/internal/Intrinsics;
at com.camerakit.CameraPreview.<init>(Unknown Source:2)
at com.camerakit.CameraKitView.onResume(CameraKitView.java:368)

This issue was not there in **com.camerakit:camerakit:1.0.0-Snapshot** and also does not come up in com.wonderkiln:camerakit:0.13.2

austinkettner, 2018-08-06:
Hey @pinigtech can you provide reproduction steps?

pinigtech, 2018-08-07:
Hi Austin,
It was probably a device specific issue and happened only on One Plus 6.
When trying to capture an image the CameraPreview would crash and one would
see a black screen. I think i had sent the logged error for it. It does not
happen in the Snapshot version but was happening in beta3.

Regards,
Piyush

Piyush Nigam | piyush.n@pinig.com | www.pinig.com
<https://mailtrack.io/trace/link/d6be614bd3f0dee5f4fc8a59db452a704a15a56a?url=http%3A%2F%2Fwww.pinig.com%2F&userId=2291577&signature=c36fddf88529b431>

*Pinig Tech Private Limited*

Noida: A-175, Sec 63, Noida 201301. Ph. +91-120-4235553

Delhi: 1UA, Jawahar Nagar,Delhi-110007

*DISCLAIMER: This email (including any attachments) is intended for the
sole use of the intended recipient/s and may contain material that is
CONFIDENTIAL COMPANY INFORMATION. Any review, reliance by others, copying,
distribution or forwarding of any or all of the contents in this message is
PROHIBITED. The opinions expressed are those of the sender, and do not
necessarily reflect those of the Company. If you are not the intended
recipient, please contact the sender by email*

pinigtech, 2018-08-07:
Ok I checked on beta3.1 now on captureImage we get an error which seems to
show that it does not find kotlin.jvm.internal.Lambda :

08-07 18:59:54.110 27847-27847/com.tams.vesselinspection I/zygote64:
Rejecting re-init on previously-failed class
java.lang.Class<com.camerakit.CameraPreview$capturePhoto$1>:
java.lang.NoClassDefFoundError: Failed resolution of:
Lkotlin/jvm/internal/Lambda;
08-07 18:59:54.110 27847-27847/com.tams.vesselinspection I/zygote64:     at
void com.camerakit.CameraKitView.obtainAttributes(android.content.Context,
android.util.AttributeSet) (CameraKitView.java:259)

pinigtech, 2018-08-07:
Added info:
This does not happen on 1.0.0-Snapshot but does happen 1.0.0-beta3.1.

Do let me know if I should be doing anything differently.

austinkettner, 2018-08-28:
Should be resolved as of Beta4, will reOpen if not!

austinkettner, 2018-10-18:
No longer seeing this in Beta3.x and forward. Closing.

closed 2018-10-18 by austinkettner