CameraKit/camerakit-android #397: Landscape images also show in portrait

pinigtech, 2018-08-06:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

One Plus 6 with Android 8.1

### Steps to Reproduce

(Write your steps here:)

1) Take one picture keeping the camera in portrait mode
2) Take another picture keeping the camera in landscape mode
3) View the pictures and you will find they will look like the ones below:
[image]

### Expected Behavior

The image on the left should have shown rotated 90 degrees in landscape ( works that way in com.wonderkiln:camerakit:0.13.2 but not in com.camerakit:camerakit:1.0.0-Snapshot)

### Actual Behavior

The screenshot above shows the image on the left is 90 degrees off - as if clicked in portrait mode only

### Reproducible Demo

Here is my code through which I get the image:

imageCallback = new CameraKitView.ImageCallback() {
            //on every click of the image from the camera
            @Override
            public void onImage(CameraKitView cameraKitView, byte[] bytes) {
                Bitmap bitmap = BitmapFactory.decodeByteArray(bytes, 0, bytes.length);

darkprize, 2018-08-09:
in my case.. using the senesor and rotate myself.
want to know how to get the rotate value from camerakitview object

austinkettner, 2018-11-19:
Should be fine in #318 please let us know if not!

closed 2018-11-19 by austinkettner