CameraKit/camerakit-android #398: Crop not cropping

pinigtech, 2018-08-07:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

One Plus 6.
Using 'com.wonderkiln:camerakit:0.13.2'
### Steps to Reproduce

(Write your steps here:)

1. Take a picture - Check Height and Width
2. use jpegTransformer.crop(Rect)
3. Check Height and Width again
4. No change in Height and Width then no crop happened:

My code:

@Override
            public void onImage(CameraKitImage cameraKitImage) {
                JpegTransformer jpegTransformer = new JpegTransformer(cameraKitImage.getJpeg());
                byte[] bytes = new byte[1];
                Rect cropBox;
                Log.i(TAG, "The width of the captured image prior cropping is " + jpegTransformer.getWidth());
                Log.i(TAG, "The height of the captured image prior cropping is " + jpegTransformer.getHeight());
                Bitmap bitmap;
                if (jpegTransformer.getWidth() < jpegTransformer.getHeight()) {
                    //Portrait
                    cropBox = new Rect(40, 331, 40, 331);
                    jpegTransformer.crop(cropBox);
                    Log.i(TAG, "The width of the captured image after cropping is " + jpegTransformer.getWidth());
                    Log.i(TAG, "The height of the captured image after cropping is " + jpegTransformer.getHeight());
                    bytes = jpegTransformer.getJpeg();
                    bitmap = BitmapFactory.decodeByteArray(bytes, 0, bytes.length);
                    //bitmap = Bitmap.createScaledBitmap(bitmap, portraitWidth, landscapeWidth, true);
                } else {
                    //Landscape
                    cropBox = new Rect(331, 40, 331, 40);
                    jpegTransformer.crop(cropBox);
                    Log.i(TAG, "The width of the captured image after cropping is " + jpegTransformer.getWidth());
                    Log.i(TAG, "The height of the captured image after cropping is " + jpegTransformer.getHeight());
                    bytes = jpegTransformer.getJpeg();
                    bitmap = BitmapFactory.decodeByteArray(bytes, 0, bytes.length);
                    //bitmap = Bitmap.createScaledBitmap(bitmap, landscapeWidth, portraitWidth, true);
                }
                Log.i(TAG, "The width of the captured image from Bitmap after cropping is " + bitmap.getWidth());
                Log.i(TAG, "The height of the captured image from Bitmap after cropping is " + bitmap.getHeight());

### Expected Behavior

I expected the logged height and width to change

### Actual Behavior

The height and width did not change :(

### Reproducible Demo

I have pasted my code above

corjohnson, 2018-08-07:
Minor tweak to the code so that it's easier to copy + paste

```Java
@override
public void onImage(CameraKitImage cameraKitImage) {
  JpegTransformer jpegTransformer = new JpegTransformer(cameraKitImage.getJpeg());
  byte[] bytes = new byte[1];
  Rect cropBox;
  Log.i("+cam", "The width of the captured image prior cropping is " + jpegTransformer.getWidth());
  Log.i("+cam", "The height of the captured image prior cropping is " + jpegTransformer.getHeight());
  Bitmap bitmap;
  if (jpegTransformer.getWidth() < jpegTransformer.getHeight()) {
    //Portrait
    cropBox = new Rect(40, 331, 40, 331);
    jpegTransformer.crop(cropBox);
    Log.i("+cam", "The width of the captured image after cropping is " + jpegTransformer.getWidth());
    Log.i("+cam", "The height of the captured image after cropping is " + jpegTransformer.getHeight());
    bytes = jpegTransformer.getJpeg();
    bitmap = BitmapFactory.decodeByteArray(bytes, 0, bytes.length);
    //bitmap = Bitmap.createScaledBitmap(bitmap, portraitWidth, landscapeWidth, true);
  } else {
    //Landscape
    cropBox = new Rect(331, 40, 331, 40);
    jpegTransformer.crop(cropBox);
    Log.i("+cam", "The width of the captured image after cropping is " + jpegTransformer.getWidth());
    Log.i("+cam", "The height of the captured image after cropping is " + jpegTransformer.getHeight());
    bytes = jpegTransformer.getJpeg();
    bitmap = BitmapFactory.decodeByteArray(bytes, 0, bytes.length);
    //bitmap = Bitmap.createScaledBitmap(bitmap, landscapeWidth, portraitWidth, true);
 }
 Log.i("+cam", "The width of the captured image from Bitmap after cropping is " + bitmap.getWidth());
 Log.i("+cam", "The height of the captured image from Bitmap after cropping is " + bitmap.getHeight());
}```

corjohnson, 2018-08-11:
Example project attached in case anyone else wants to try to repro! I was able to reproduce this issue as well. 

[CameraKitTest.zip](https://github.com/CameraKit/camerakit-android/files/2279798/CameraKitTest.zip)

pinigtech, 2018-08-11:
Thanks for sharing this Corey. I will take a look at this and come back to
you with what I find. To make it work on One Plus 6 phone here is the
combination of thigs that failed and worked:
1) com.camerakit:camerakit:1.0.0-beta3.1 does not show a camera preview so
failed
2) com.wonderkiln.camerakit:0.13.2 worked mostly well but had two issues in
my limited requirement:

a) in landscape view, after a click the preview used to get distorted

b) The crop did not work as stated at the beginning of this ticket

The workarounds I have made which solved it for me (and this might be
specific to OP6 only) - for solving (a) I downgraded to
com.wonderkiln.camerakit:0.13.0
and for solving (b) I replaced the com.wonderkiln.JpegTrasformer with
jpegkit.JpegTransformer
.
So now the above might just be a group of workarounds but it does the job
for me today when I ran out of time to wait for the perfect 1.0.0 to come
up...

emersoncloud, 2019-01-29:
Hey @pinigtech we've recently updated [JpegKit](https://github.com/CameraKit/jpegkit-android) to version `v0.2.2`. Try to crop again with the updated version. We recommend  you use `Jpeg` to crop rather than `JpegTransformer`. 

Closing this issue. If the problem persists I will reopen. Thanks!

closed 2019-01-29 by emersoncloud