CameraKit/camerakit-android #401: Finishing activity immediately after captureImage() crashes the app.

pawarprasad37, 2018-08-11:
### Is this a bug report?
Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?
Yes

### Steps to Reproduce
(The camera activity has CameraKit onResume and onPause invocations in its relevant lifecycle methods).
1. Invoke cameraKitView.captureImage(...)
2. Finish the activity in 500 milli-seconds using Handler.postDelayed(). 
(This is the simulate user behavior of killing the app after clicking capture button on UI and before CameraKit gives image callback).
3. App will crash.

### Expected Behavior
App should not crash and there should be no callback.
If camera is released immediately after captureImage() invocation, the capture procdess should be stopped silently.

### Actual Behavior
App crashes with the below logs.

java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.graphics.Bitmap.compress(android.graphics.Bitmap$CompressFormat, int, java.io.OutputStream)' on a null object reference
        at com.camerakit.CameraKitView$Camera2$1$7.run(CameraKitView.java:1855)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)

### Reproducible Demo
https://github.com/pawarprasad37/camerakitbackpressbugrepro

emersoncloud, 2018-11-19:
@pawarprasad37 Thanks for the issue and the demo application, it was very helpful!

I was able to reproduce the issue on CameraKit version 1.0.0-snapshot like you have above. However switching to version 1.0.0-beta3.9 CameraKit no longer crashes during the capture image process. Let me know if you run into any more issues.

closed 2018-11-19 by emersoncloud

emersoncloud, 2018-11-19:
As a side note implementation instructions for beta3.9 are up on the main README here: https://github.com/CameraKit/camerakit-android