CameraKit/camerakit-android #404: AndroidX Support

Vincent-Carrier, 2018-08-21:
### Environment

Android Studio 3.2 Beta 5

### Steps to Reproduce

1. Include the CameraKit library with AndroidX libraries (which will be default in AS 3.2). 
2. Build the project.

### Expected Behavior

The project builds properly

### Actual Behavior

Build fails with this message

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForDebug'.
> com.android.builder.dexing.DexArchiveMergerException: Error while merging dex archives: /Users/vincent/AndroidStudioProjects/FlashDiet/app/build/intermediates/transforms/dexBuilder/debug/3.jar, /Users/vincent/AndroidStudioProjects/FlashDiet/app/build/intermediates/transforms/dexBuilder/debug/7.jar, 
...
/Users/vincent/AndroidStudioProjects/FlashDiet/app/build/intermediates/transforms/dexBuilder/debug/374.jar, /Users/vincent/AndroidStudioProjects/FlashDiet/app/build/intermediates/transforms/dexBuilder/debug/375.jar
  Program type already present: android.support.v4.os.ResultReceiver
  Learn how to resolve the issue at https://developer.android.com/studio/build/dependencies#duplicate_classes.
```

### Reproducible Demo

https://github.com/Vincent-Carrier/FlashDiet/tree/965a8e50a798bd86b7b0b0f60f03e0da5b328d28

### Solution

Publish a branch with AndroidX dependencies. AS 3.2 has a 'Refactor to AndroidX' option.

See `camerakit-android/camerakit-core/build.gradle`, lines 37 and 38

emersoncloud, 2018-08-24:
@Vincent-Carrier Thanks for the suggestion. We've talked internally about AndroidX support. I'll put this on our 1.1 roadmap and keep you updated on the progress.

emersoncloud, 2019-01-29:
We've chosen not to include AndroidX in the latest CameraKit beta versions, at least for the time being. Adding AndroidX required that the end user including CameraKit also needed to include AndoridX in their application. So in an effort to keep CameraKit as widely usable as possible, we've left it out.

Thanks again for the suggestion!

closed 2019-01-29 by emersoncloud

eygraber, 2019-03-29:
Please update to use androidx dependencies. This library is the only thing preventing us from turning off jetifier. Or at least offer an androidx version.