CameraKit/camerakit-android #405: distorted in preview

suarezu, 2018-08-22:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment
Xiaomi mi8
Android 8.1.0 API 27

### Steps to Reproduce

(Write your steps here:)

1. add a camera view in some layout
<ConstraintLayout>
    <com.wonderkiln.camerakit.CameraView
        android:id="@+id/camera"
        android:layout_width="0dp"
        android:layout_height="0dp"
        android:adjustViewBounds="true"
        app:ckLockVideoAspectRatio="true"
        app:ckVideoQuality="max1080p"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintHeight_default="percent"
        app:layout_constraintHeight_percent=".8"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent"
        app:layout_constraintVertical_bias="0" />
</ConstraintLayout>

### Expected Behavior

the preview should look as what it is

### Actual Behavior

but the preview seems distorted, the ratio doesnt make sense, the screenshots as attached
[image]
[image]

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

suarezu, 2018-08-22:
`
    <com.wonderkiln.camerakit.CameraView
        android:id="@+id/camera"
        android:layout_width="0dp"
        android:layout_height="0dp"
        android:adjustViewBounds="true"
        app:ckLockVideoAspectRatio="true"
        app:ckVideoQuality="max1080p"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintHeight_default="percent"
        app:layout_constraintHeight_percent=".8"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent"
        app:layout_constraintVertical_bias="0" />`

this is the code in a ConstraintLayout, and ckVideoQuality takes no effect, since the logged preview size 1920x1440 and capture size 3024x4032 are not as fed.

suarezu, 2018-08-22:
the library version is 0.13.2, I found a solution which respects the video quality set by the user to the bug:

`
private TreeSet<AspectRatio> findCommonAspectRatios(List<Camera.Size> previewSizes, List<Camera.Size> pictureSizes) {
        Set<AspectRatio> previewAspectRatios = new HashSet<>();
        CamcorderProfile profile = getCamcorderProfile(mVideoQuality);
        AspectRatio userAspectRatio = AspectRatio.of(profile.videoFrameWidth, profile.videoFrameHeight);
        for (Camera.Size size : previewSizes) {
            AspectRatio previewRatio = AspectRatio.of(size.width, size.height);
            if (userAspectRatio.equals(previewRatio)) {
                previewAspectRatios.add(previewRatio);
            }
        }

        Set<AspectRatio> captureAspectRatios = new HashSet<>();
        for (Camera.Size size : pictureSizes) {
            captureAspectRatios.add(AspectRatio.of(size.width, size.height));
        }

        TreeSet<AspectRatio> output = new TreeSet<>();
        if (previewAspectRatios.size() == 0) {
            // if no common aspect ratios
            Camera.Size maxSize = previewSizes.get(0);
            AspectRatio maxPreviewAspectRatio = AspectRatio.of(maxSize.width, maxSize.height);
            for (AspectRatio aspectRatio : captureAspectRatios) {
                if (aspectRatio.equals(maxPreviewAspectRatio)) {
                    output.add(aspectRatio);
                }
            }
        } else {
            // if common aspect ratios exist
            for (AspectRatio aspectRatio : previewAspectRatios) {
                if (captureAspectRatios.contains(aspectRatio)) {
                    output.add(aspectRatio);
                }
            }
        }

        return output;
    }
`

sorry no idea of how to contribute, and I want to upgrade to v1.0.0 to see whether this problem ramins in the latest version, but gradle build fails:
`Failed to resolve: com.camerakit:camerakit:1.0.0`

emersoncloud, 2018-08-24:
Hi @suarezu we've got this issue documented here #384, check back there for updates. Implementation details of how to use 1.0.0 can be found on the PR #318.

closed 2018-08-24 by emersoncloud