CameraKit/camerakit-android #412: Flipped Video Recorded on Samsung J5 Prime

santoshthorani, 2018-09-06:
### Is this a bug report?

(write your answer here)

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

(Write your answer here.)

### Environment

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

emersoncloud, 2018-09-11:
@santoshthorani which version of CameraKit are you using?

emersoncloud, 2018-11-14:
Closing this issue. If you're still experiencing issues comment back and I can re-open the issue.

closed 2018-11-14 by emersoncloud