CameraKit/camerakit-android #414: Crash on start

deeroad, 2018-09-07:
I change my CameraKit version from 1.0.0-snapchat to 1.0.0-beta3.9,and my app crashed whenever it started. It runs well on version snapchat.
The logcat prints the follwing errors:
------------------------------------------------------
09-07 15:36:03.189 26200-26200/com.alasoft.photoclient E/MemoryLeakMonitorManager: MemoryLeakMonitor.jar is not exist!
09-07 15:36:03.189 26200-26200/com.alasoft.photoclient E/Minikin: Could not get cmap table size!
09-07 15:36:03.274 26200-26200/com.alasoft.photoclient E/xuquanfei: xuquanfei_getCallerProcessName is used callingApp = com.alasoft.photoclient
09-07 15:36:03.523 26200-26200/com.alasoft.photoclient E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.alasoft.photoclient, PID: 26200
    android.view.InflateException: Binary XML file line #9: Binary XML file line #9: Error inflating class com.camerakit.CameraKitView
    Caused by: android.view.InflateException: Binary XML file line #9: Error inflating class com.camerakit.CameraKitView
    Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:334)
        at android.view.LayoutInflater.createView(LayoutInflater.java:658)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:801)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:741)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:874)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:835)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:515)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:423)
        at com.alasoft.photoclient.common.BaseFragment.onCreateView(BaseFragment.java:28)
        at android.support.v4.app.Fragment.performCreateView(Fragment.java:2346)
        at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1428)
        at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1759)
        at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1827)
        at android.support.v4.app.BackStackRecord.executeOps(BackStackRecord.java:797)
        at android.support.v4.app.FragmentManagerImpl.executeOps(FragmentManager.java:2596)
        at android.support.v4.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2383)
        at android.support.v4.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2338)
        at android.support.v4.app.FragmentManagerImpl.execSingleAction(FragmentManager.java:2215)
        at android.support.v4.app.BackStackRecord.commitNowAllowingStateLoss(BackStackRecord.java:649)
        at android.support.v4.app.FragmentPagerAdapter.finishUpdate(FragmentPagerAdapter.java:145)
        at android.support.v4.view.ViewPager.populate(ViewPager.java:1238)
        at android.support.v4.view.ViewPager.populate(ViewPager.java:1086)
        at android.support.v4.view.ViewPager.onMeasure(ViewPager.java:1616)
        at android.view.View.measure(View.java:22289)
        at android.support.constraint.ConstraintLayout.internalMeasureChildren(ConstraintLayout.java:934)
        at android.support.constraint.ConstraintLayout.onMeasure(ConstraintLayout.java:973)
        at android.view.View.measure(View.java:22289)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6668)
        at android.widget.FrameLayout.onMeasure(FrameLayout.java:185)
        at android.support.v7.widget.ContentFrameLayout.onMeasure(ContentFrameLayout.java:141)
        at android.view.View.measure(View.java:22289)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6668)
        at android.support.v7.widget.ActionBarOverlayLayout.onMeasure(ActionBarOverlayLayout.java:400)
        at android.view.View.measure(View.java:22289)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6668)
        at android.widget.FrameLayout.onMeasure(FrameLayout.java:185)
        at android.view.View.measure(View.java:22289)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6668)
        at android.widget.LinearLayout.measureChildBeforeLayout(LinearLayout.java:1539)
        at android.widget.LinearLayout.measureVertical(LinearLayout.java:823)
        at android.widget.LinearLayout.onMeasure(LinearLayout.java:702)
        at android.view.View.measure(View.java:22289)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6668)
        at android.widget.FrameLayout.onMeasure(FrameLayout.java:185)
        at com.android.internal.policy.DecorView.onMeasure(DecorView.java:834)
        at android.view.View.measure(View.java:22289)
        at android.view.ViewRootImpl.performMeasure(ViewRootImpl.java:2609)
        at android.view.ViewRootImpl.measureHierarchy(ViewRootImpl.java:1641)
        at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:1900)
09-07 15:36:03.524 26200-26200/com.alasoft.photoclient E/AndroidRuntime:     at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1525)
        at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7319)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:981)
        at android.view.Choreographer.doCallbacks(Choreographer.java:790)
        at android.view.Choreographer.doFrame(Choreographer.java:721)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:967)
        at android.os.Handler.handleCallback(Handler.java:809)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:166)
[... 12 lines cut ...]
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        	... 66 more
    	Suppressed: java.io.IOException: No original dex files found for dex location /data/app/com.alasoft.photoclient-Nk6RSaDKkbfkSPLyEkPRDw==/split_lib_resources_apk.apk
        at dalvik.system.DexFile.openDexFileNative(Native Method)
        at dalvik.system.DexFile.openDexFile(DexFile.java:353)
        at dalvik.system.DexFile.<init>(DexFile.java:100)
        at dalvik.system.DexFile.<init>(DexFile.java:74)
        at dalvik.system.DexPathList.loadDexFile(DexPathList.java:374)
        at dalvik.system.DexPathList.makeDexElements(DexPathList.java:337)
        at dalvik.system.DexPathList.<init>(DexPathList.java:157)
        at dalvik.system.BaseDexClassLoader.<init>(BaseDexClassLoader.java:65)
        at dalvik.system.PathClassLoader.<init>(PathClassLoader.java:64)
        at com.android.internal.os.ClassLoaderFactory.createClassLoader(ClassLoaderFactory.java:73)
        at com.android.internal.os.ClassLoaderFactory.createClassLoader(ClassLoaderFactory.java:88)
        at android.app.ApplicationLoaders.getClassLoader(ApplicationLoaders.java:75)
        at android.app.ApplicationLoaders.getClassLoader(ApplicationLoaders.java:37)
        at android.app.LoadedApk.createOrUpdateClassLoaderLocked(LoadedApk.java:732)
        at android.app.LoadedApk.getClassLoader(LoadedApk.java:766)
        at android.app.LoadedApk.getResources(LoadedApk.java:1013)
        at android.app.ContextImpl.createAppContext(ContextImpl.java:2537)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6360)
        at android.app.ActivityThread.-wrap2(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1969)
        at android.os.Handler.dispatchMessage(Handler.java:109)
        		... 5 more

emersoncloud, 2018-09-11:
@deeroad you need to add two new lines to the dependencies section of your `build.gradle` file. 

```
dependencies {
    // ...
    implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.2.61'
    implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:0.24.0'
}
```

Check out the PR comments #318 for more information. Let me know if you're still having issues!

closed 2018-09-11 by emersoncloud

zpymyyn, 2018-10-29:
> @deeroad you need to add two new lines to the dependencies section of your `build.gradle` file.
>
> ```
> dependencies {
>     // ...
>     implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.2.61' [...]

Hi there,

I still met with the issue after adding those lines...
Appreciate any help.
Phone in use: Huawei P20 Lite

austinkettner, 2018-10-29:
Hi @zpymyyn can you please share your full gradle implementation and a code sample of what you are trying to do?

HemanthImandi, 2019-07-30:
I added those four lines in gradle i got the "Error inflating class com.camerakit.CameraKitView".
Is there any solution for this.