CameraKit/camerakit-android #415: android.graphics.SurfaceTexture.setDefaultBufferSize(int, int) crash

frontiertsymbal, 2018-09-10:
Hi. Thanks for great library.

I have a crash on Meizu M5s after taking photo. I haven`t  more info, because the app is crashed on device 1100 km away from me.

Log from Crashlytics:

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference
       at com.camerakit.CameraKitView$Camera2$1$3.run(CameraKitView.java:1737)
       at android.os.Handler.handleCallback(Handler.java:815)
       at android.os.Handler.dispatchMessage(Handler.java:104)
       at android.os.Looper.loop(Looper.java:207)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

emersoncloud, 2018-09-11:
Hi @frontiertsymbal could you test on the latest beta, v1.0.0-beta3.9? There are a couple implementation differences for beta3.9, you can see those in the PR comments #318. 

Are you experiencing this on other devices or just that one Meizu M5s? Which version of Android is that device running? Thanks!

frontiertsymbal, 2018-09-12:
Xiaomy Redmi Note 4 - Android 7.0
Meizu M5s - Android 6.0

On Huawei Nexus 6p (Android 8.1) and Xiaomi A1 (Android 8.1) and LG Nexus 5 (Android 6.0.1) all works fine. I haven`t access to Xiaomy Redmi Note 4 and Meizu M5s, only Crashlitycs report.

Today i`ve tested v1.0.0-beta3.9 - i have another problem with native code with very very long stacktrace - looks like something wrong with Camera initialization with DataBinding. 

StackTrace on GDrive - https://drive.google.com/file/d/1vLIU89_Mf1I1IlycSkxc9OEM1pWITW_k/view?usp=sharing , devices - Xiaomi A1 and Huawei Nexus 6p.

frontiertsymbal, 2018-09-16:
@emersoncloud Hi. Any updates regarding these bugs?

BenMoveo, 2018-11-08:
hi @emersoncloud the crash reported here happens to me also no matter on which device - (Pixel emulator, Huawei P9, LG K10)

I noticed it happens when I start the app again (from my IDE) without shutting down the app.
I've tried to test with v1.0.0-beta3.9 but the problem remains.
Waiting for updates,
Ben.

emersoncloud, 2018-11-14:
Hi @BenMoveo can you give me some more context about the crash you're experiencing? 

Are you getting the same error on both 1.0.0-snapshot as you are on 1.0.0-beta3.9?

A log of the error would be very helpful as well as your implementation of CameraKit in the XML and Java. Thanks!

emersoncloud, 2018-11-14:
@frontiertsymbal Could you try v1.0.0-beta3.9 following the instruction on the README here: https://github.com/CameraKit/camerakit-android.

emersoncloud, 2019-01-29:
@frontiertsymbal any updates on errors on your end? Closing this issue for now but will reopen if problem persists!

closed 2019-01-29 by emersoncloud