CameraKit/camerakit-android #416: Documentation is hopelessly out of date

bpappin, 2018-09-10:
The documentation is so out of date, its pretty useless.
See:
http://docs.camerakit.website/#/

austinkettner, 2018-09-10:
Rest assured it's not hopeless!

It's a known area we need to improve vastly, we are waiting on Beta4 with it's final API. After that we will be rapidly improving the documentation around v1.0.0.

The documentation is entirely aimed at 0.13.x and for that release it is up to date!

ArslanKathia, 2018-09-28:
when the new Version of camera-Kit is released?

austinkettner, 2018-10-29:
Hey all, we expect to be rolling out a new documentation site and proper documentation for the various current releases of CameraKit within the next two weeks.

closed 2018-10-29 by austinkettner

usmanrana07, 2019-04-22:
@austinkettner  when is new documentation or new major version expected as few months in wait.