CameraKit/camerakit-android #419: How to use getFrames version 1.0.0

EdwardTeac, 2018-09-13:
### Is this a bug report?

Yes

### Have you read the [Contributing Guidelines](.github/CONTRIBUTING.md)?

Yes

### Environment

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

Im using the following code to catch frames, and share camera preview
cameraKitView.captureFrame(new CameraKitView.FrameCallback() {
                    @Override
                    public void onFrame(CameraKitView cameraKitView, byte[] bytes) {

                        bitmap = BitmapFactory.decodeByteArray(bytes , 0, bytes.length);
                        cameraFrames.setImageBitmap(bitmap);
                    }
                });

But it doesn display anything :(

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

emersoncloud, 2018-11-14:
Like @XinyueZ mentioned, please follow up on ticket #342 for frame inspection updates. Closing this ticket.

closed 2018-11-14 by emersoncloud

johcam12, 2019-03-15:
@emersoncloud #342 points back to this issue (https://github.com/CameraKit/camerakit-android/issues/342#issuecomment-424252674) which makes this issue non GC-collectable.

RahulRawat95, 2019-04-09:
Is this implemented yet?

rohfosho, 2019-05-04:
@dwillmc Any updates on when `captureFrame` will be implemented?

Zaniyar, 2019-05-26:
@rohfosho I don't think it will ever be implemented.