CameraKit/camerakit-android #426: RFE - need to be able to check if the flash is present.

bpappin, 2018-09-21:
This is an RFE rather than a bug report.

I'm finding a lot of devices don't have a flash. 
It would be very handy if I could find out if i should show the flash button with a simple call to `<cameraview>.hasFlash()`.

Some information about how this is done can be found in this Stack Overflow link.
https://stackoverflow.com/questions/13413938/how-to-check-if-device-has-flash-light-led-android

At this point, the codebase seems to be a little out of sync, with a pending 1.0.0 merge request, so I don't want to add another merge request on top of that.
However, if the main team can get the merge requests cleared up, I don't mind putting in the work, to get this RFE done.

austinkettner, 2018-09-21:
Thanks for the RFE @bpappin - we love getting tickets like this!

I have marked this as something to be sure to include in our finalized 1.0.0 release. You are correct that the current v1.0.0 is causing a few things to be out of sync, we are working hard to get everything back in line.

bpappin, 2018-09-21:
You might also include a hasCamera(FRONT|REAR), as sometimes a device doesn't have one or the other.

Or maybe it would be more flexible to be able to access the camera device object. I could then have done the flash check myself :)

emersoncloud, 2019-01-29:
Hey @bpappin we've added the `hasFlash()` method in CameraKit `v1.0.0-beta3.11`. Check out the docs here: https://camerakit.io/docs.

I've created a new ticket for the hasCamera request here: #497. I'll post updates there as they come!

PR #474 (2019-01-23): v1.0.0-beta3.11
## CameraKit v1.0.0-beta3.11

**Issues covered in this release:**
Closes #348
Closes #426 
Closes #479
closed 2019-01-29 by emersoncloud