CameraKit/camerakit-android #427: 1.0.0 Beta 3.9 - Camera preview not rotating correctly with device orientation

MeteC, 2018-09-25:
### Environment

Development:
- Android Studio 3.2 (Also found in AS 3.1), Mac High Sierra

Running Devices:
- Xiaomi A1 (Android 8.1)
- OnePlus 3T (Android 8.0)

### Steps to Reproduce

See attached project, the gist is:

1. New Android project
2. Include CameraKit beta 3.9 in gradle 
3. Create a basic CameraKit Activity which just shows a camera preview on screen
4. Fire up the app on one of the above running devices, in portrait mode
5. Now rotate to landscape mode

Alternatively, launch the app in landscape mode, then rotate to portrait mode.

### Expected Behavior

Camera preview should rotate correctly with screen orientation

### Actual Behavior

The screen orientation changes, the camera preview rotates incorrectly, and stays 90 degrees out.

This does *not* happen for me on the following devices (i.e. they look perfect):

- Moto G5 (Android 7.0)
- Samsung A5 (Android 6.0.1)

### Reproducible Demo

[CameraKitTester AS Project.zip](https://github.com/CameraKit/camerakit-android/files/2414687/CameraKitTester.AS.Project.zip)

### Video Demonstration (what I'm seeing)

[image]

MeteC, 2018-09-25:
#318

emersoncloud, 2018-09-25:
Much thanks for the detailed issue! I was able to reproduce it on my test devices and got similar results that you mentioned.

Has rotation issue on:
* Pixel XL - Android 9.0
* Moto G6 - Android 8.0

Issue does not happen on:
* Samsung Galaxy S7 - Android 7.0

I was able to fix the issue entirely by moving the `CameraKitView` out of the `CameraKitActivity.java` and into the `MainActivity.java`. It seems that creating and calling the separate activity is what's causing the rotation issues. I'm still investigating why, I'll keep you updated!

MeteC, 2018-09-26:
Ah good to hear, thanks!

KlaartjeD, 2018-10-19:
Any news on this? 
I was able reproduce this on the Samsung Galaxy Tab S3

austinkettner, 2018-10-26:
Looking to resolve in a upcoming beta release!

MeteC, 2018-11-30:
ok cool. I can't remove that line from my project but good to know we're getting closer :-) 👍

emersoncloud, 2019-01-30:
Closing this issue in favor of #384. I'll make a note of it there. You can watch for updates on that ticket. Thanks again for bringing this to our attention!

closed 2019-01-30 by emersoncloud

HBiSoft, 2019-05-21:
@emersoncloud any updates on this? I commented on #384 describing the issue. Are you planning to fix this?

ErAmrit, 2019-06-18:
Camera Preview Rotation Issue: When Rotating from Landscape to Reverse Landscape preview is not correct and it rotated to 180 degree, while when rotating Portrait ->LandScape->Reverse Portrait->Reverse LandScape or vice-versa then Preview is fine. 
i am using version 1.0.0-beta3.1, also tried 1.0.0-beta3.11, 3.10 and beta4 but these version more venerable while  version 1.0.0-beta3.1 is stable to the some extend only i am getting this "Landscape to Reverse Landscape" preview is not correct.